platformio/platform-espressif32 #218: COMPONENT_EMBED_FILES not working in library.json

Dee-Fuse, 2019-08-09:
I'm trying to add binary data to the firmware as in #161 but bundled into an library.

I've added the build flag but no matter what I try to change it does not find / include the corresponding files.

Is it possible to include such a flag into a library or do I have to include it into the platformio.ini?

Dee-Fuse, 2019-08-21:
Its looks like the build options from an library.json file are not passed through `builder/frameworks/_embedtxt_files.py` to embedd files. The problem may lay in the platformio-core instead of the esp32 platform package.

valeros, 2019-08-21:
Hi @Dee-Fuse! Have you tried [extra_script](http://docs.platformio.org/en/latest/librarymanager/config.html#extrascript) option?

Dee-Fuse, 2019-08-21:
Yes, I've tryed it but without luck.
The problem with that option is, that the `extra_script` is been executed after `_emebdtxt_files.py`.
Due to that behaviour, the adding of the `CPPDEFINES` flags here never gets picked up by the embedding script.

I've tried:
```
Import('env')

env.Append(
    CPPDEFINES=["-D COMPONENT_EMBED_FILES www/style.css:www/jquery.gz:www/code.js:www/index.html"]
)
```
And 
```
global_env = DefaultEnvironment()
global_env.Append(
    CPPDEFINES=["-D COMPONENT_EMBED_FILES www/style.css:www/jquery.gz:www/code.js:www/index.html"]
)
```

Here is the output of the build process: ( I've added a print(env.get("CPPDEFINES")) in both files to verify it)
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30202.190627 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)
[('PLATFORMIO', 40000), 'ARDUINO_ESP32_DEV', ('CONFIG_WIFI_MANAGER_TASK_PRIORITY', 5), ('CONFIG_WIFI_MANAGER_MAX_RETRY', 2), ('CONFIG_DEFAULT_AP_SSID', '\\"esp32\\"'), ('CONFIG_DEFAULT_AP_PASSWORD', '\\"esp32pwd\\"'), ('CONFIG_DEFAULT_AP_CHANNEL', 1), ('CONFIG_DEFAULT_AP_IP', '\\"10.10.0.1\\"'), ('CONFIG_DEFAULT_AP_GATEWAY', '\\"10.10.0.1\\"'), ('CONFIG_DEFAULT_AP_NETMASK', '\\"255.255.255.0\\"'), ('CONFIG_DEFAULT_AP_MAX_CONNECTIONS', 4), ('CONFIG_DEFAULT_AP_BEACON_INTERVAL', 100), 'ESP32', 'ESP_PLATFORM', ('F_CPU', '$BOARD_F_CPU'), 'HAVE_CONFIG_H', ('MBEDTLS_CONFIG_FILE', '\\"mbedtls/esp_config.h\\"')]
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
InExtraScript:
[('WITH_POSIX',), ('IDF_VER', '\\"3.30202.190627\\"'), ('PLATFORMIO', 40000), ('ARDUINO_ESP32_DEV',), ('CONFIG_WIFI_MANAGER_TASK_PRIORITY', 5), ('CONFIG_WIFI_MANAGER_MAX_RETRY', 2), ('CONFIG_DEFAULT_AP_SSID', '\\"esp32\\"'), ('CONFIG_DEFAULT_AP_PASSWORD', '\\"esp32pwd\\"'), ('CONFIG_DEFAULT_AP_CHANNEL', 1), ('CONFIG_DEFAULT_AP_IP', '\\"10.10.0.1\\"'), ('CONFIG_DEFAULT_AP_GATEWAY', '\\"10.10.0.1\\"'), ('CONFIG_DEFAULT_AP_NETMASK', '\\"255.255.255.0\\"'), ('CONFIG_DEFAULT_AP_MAX_CONNECTIONS', 4), ('CONFIG_DEFAULT_AP_BEACON_INTERVAL', 100), ('ESP32',), ('ESP_PLATFORM',), ('F_CPU', '$BOARD_F_CPU'), ('HAVE_CONFIG_H',), ('MBEDTLS_CONFIG_FILE', '\\"mbedtls/esp_config.h\\"'), ('GCC_NOT_5_2_0', 0), '-D COMPONENT_EMBED_FILES $PROJECT_DIR/www/style.css:www/jquery.gz:www/code.js:www/index.html']
```

valeros, 2019-10-23:
Sorry for the late reply. But at the moment it's not possible to detect `CPPDEFINES` from library extra script when the build script is being processed. As a workaround you could extract our script `embed_files.py` and add it to your library. So it should look something like this:
`extra_script.py`:
```
import shutil
from os import SEEK_CUR, SEEK_END
from os.path import basename, isfile, join

from SCons.Script import Builder

from platformio.util import cd

Import("env")

global_env = DefaultEnvironment()

#
# Embedded files helpers
#

def extract_files(cppdefines, files_type):
    for define in cppdefines:
        if files_type not in define:
            continue

        if not isinstance(define, tuple):
            print("Warning! %s macro cannot be empty!" % files_type)
            return []

        with cd(env.subst("$PROJECT_DIR")):
            value = define[1]
            if not isinstance(value, str):
                print("Warning! %s macro must contain "
                      "a list of files separated by ':'" % files_type)
                return []

            result = []
            for f in value.split(':'):
                if not isfile(f):
                    print("Warning! Could not find file %s" % f)
                    continue
                result.append(join("$PROJECT_DIR", f))

            return result

def remove_config_define(cppdefines, files_type):
    for define in cppdefines:
        if files_type in define:
            env.ProcessUnFlags("-D%s" % "=".join(str(d) for d in define))
            return

def prepare_file(source, target, env):
    filepath = source[0].get_abspath()
    shutil.copy(filepath, filepath + ".piobkp")

    with open(filepath, "rb+") as fp:
        fp.seek(-1, SEEK_END)
        if fp.read(1) != '\0':
            fp.seek(0, SEEK_CUR)
            fp.write(b'\0')

def revert_original_file(source, target, env):
    filepath = source[0].get_abspath()
    if isfile(filepath + ".piobkp"):
        shutil.move(filepath + ".piobkp", filepath)

def embed_files(files, files_type):
    for f in files:
        filename = basename(f) + ".txt.o"
        file_target = env.TxtToBin(join("$BUILD_DIR", filename), f)
        env.Depends("$PIOMAINPROG", file_target)
        if files_type == "COMPONENT_EMBED_TXTFILES":
            env.AddPreAction(file_target, prepare_file)
            env.AddPostAction(file_target, revert_original_file)
        global_env.Append(PIOBUILDFILES=[env.File(join("$BUILD_DIR", filename))])

env.Append(
    BUILDERS=dict(
        TxtToBin=Builder(
            action=env.VerboseAction(" ".join([
                "xtensa-esp32-elf-objcopy",
                "--input-target", "binary",
                "--output-target", "elf32-xtensa-le",
                "--binary-architecture", "xtensa",
                "--rename-section", ".data=.rodata.embedded",
                "$SOURCE", "$TARGET"
            ]), "Converting $TARGET"),
            suffix=".txt.o"))
)

flags = env.get("CPPDEFINES")
for files_type in ("COMPONENT_EMBED_TXTFILES_LIB", "COMPONENT_EMBED_FILES_LIB"):
    if files_type not in env.Flatten(flags):
        continue
    files = extract_files(flags, files_type)
    embed_files(files, files_type)
    remove_config_define(flags, files_type)
```
`library.json`:
```
{
    "name": "SomeLib",
    "version": "0.0.0",
    "build": {
        "extraScript": "extra_script.py",
        "flags": "-DCOMPONENT_EMBED_TXTFILES_LIB=src/private.pem.key:src/certificate.pem.crt:src/aws-root-ca.pem "
    }
}
```
WDYT?

valeros, 2020-05-25:
Feel free to reopen the issue if you still need any help.

closed 2020-05-25 by valeros