platformio/platform-espressif32 #219: Board Support: TTGO T-Call support

Skipper-is, 2019-08-17:
- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

The TTGO T-Call is an ESP32 WROVER-B based board, featuring a SIM800L GSM/2G chip.

[T-Call on AliExpress](https://www.aliexpress.com/item/33045221960.html)

[T-Call on BangGood](https://www.banggood.com/LILYGO-TTGO-T-Call-V1_3-ESP32-Wireless-Module-GPRS-Antenna-SIM-Card-SIM800L-Board-p-1527048.html?rmmds=search&cur_warehouse=CN)

Features:

- 4mb QSPI Flash
- 8mb PSRAM
-40Mhz clock
-SIM800L chip on pins 4, 5, 23, 26 and 27.
-LiPo charger/power management
- IP5306.. I'm not sure, the data sheet is in Chinese, but I  think it is a buck converter... Outputs 2A anyway, for the SIM800L during transmission bursts (and because the SIM800L works on 3.4-4.4v)

I have used the attatched code in my ttgo-t-call.json, and it appears to work, but maybe I've made a mistake...

[pins_aruduino.h](https://github.com/platformio/platformio-core/files/3512008/pins_aruduino.h.txt)
[ttgo-t-call.json](https://github.com/platformio/platformio-core/files/3512009/ttgo-t-call.json.txt)

valeros, 2019-09-16:
Please open an issue here https://github.com/espressif/arduino-esp32/issues in the first place, so the board will be added to the Arduino framework officially.

ivankravets, 2020-04-25:
@MaksimKlyu please keep us update on this issue.

SunboX, 2021-11-30:
@MaksimKlyu the issue https://github.com/platformio/platform-espressif32/issues/333 was closed. I also need the arduino support for the ttgo-t-call. What's the current state?