platformio/platform-espressif32 #221: Invalid error message when debugging

manuelbl, 2019-08-23:
About a month ago, the below `platform.ini` was sufficient to debug an ESP32 (Arduino and ESP-IDF) with the Espressif *Esp-Prog*:

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
```

The same file now results in the below error:

```
user$ pio debug
Error: Please specify `debug_port` for environment
```

Adding `debug_port` doesn't help and results in exactly the same error. Furthermore, it would be nonsense to add `debug_port` as the Esp-Prog adapter is located via a list of USB VID/PID specified in an OpenOCD interface definition.

However, the below `platform.ini` works:

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

debug_tool = esp-prog
```

So there seem to be two problems:
- The default for `debug_tool` was lost (or is otherwise broken). According to the [board documentation](https://docs.platformio.org/en/latest/boards/espressif32/esp32doit-devkit-v1.html), it should be `esp-prog`.
- The error message is invalid. If anything, it should say: *Please specify \`debug_tool\`*

manuelbl, 2019-08-23:
I forgot to add: Currently, I'm running PIO Core 4.0.1.

ivankravets, 2019-09-18:
It works in the latest PIO Core
```
pio debug  --interface=gdb -x .pioinit
Reading symbols from /private/tmp/pio-workspaces/esp32 arduino-45409e953f/build/esp32doit-devkit-v1/firmware.elf...done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = esp-prog
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.10.0-esp32-20190708 (2019-07-08-11:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
Info : tcl server disabled
Info : telnet server disabled
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'

Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of irom does not support free_driver_priv()
Warn : Flash driver of drom does not support free_driver_priv()
.pioinit:11: Error in sourced command file:
Remote connection closed
```

Please reopen if you still have this issue.

closed 2019-09-18 by ivankravets