platformio/platform-espressif32 #226: ESPIDF v3.3 requires "pyparsing"

atanisoft, 2019-09-16:
Using latest develop branch the following is observed while building:
```
generate_project_ld_script([".pio\build\pcb_oled_sh1106\esp32.project.ld"], ....... 
Traceback (most recent call last):
  File ".....platformio\packages\framework-espidf\tools\ldgen\ldgen.py", line 22, in <module>
    from fragments import FragmentFileModel
  File "......platformio\packages\framework-espidf\tools\ldgen\fragments.py", line 20, in <module>
    from sdkconfig import SDKConfig
  File ".....platformio\packages\framework-espidf\tools\ldgen\sdkconfig.py", line 18, in <module>
    from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ImportError: No module named pyparsing
```
(above has been pruned to show only the required pieces)

After resolving the above by running "pip install pyparsing" in the penv/Scripts directory the following is observed:
```
IOError: C:\Users\mdunston\.platformio\packages\framework-espidf\Kconfig:59: Could not open 'C:\Users\mdunston\.platformio\packages\framework-espidf\components\app_update\Kconfig.projbuild#C:\Users\mdunston\.platformio\packages\framework-espidf\components\bootloader\Kconfig.projbuild#C:\Users\mdunston\.platformio\packages\framework-espidf\components\esptool_py\Kconfig.projbuild#C:\Users\mdunston\.platformio\packages\framework-espidf\components\partition_table\Kconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance
is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have
'option env="FOO"' in its definition).
```

The above is due to Windows not defining IFS in the env and the default setting being " \t\n" (space, tab, newline) inside kconfiglib.py.

Using space instead of # resolves this issue and building is successful.

@valeros FYI...

valeros, 2019-09-16:
Thanks for reporting the problem! Strange, but our CI didn't catch that. Are you sure you're using the `esp-idf` package shipped with PlatformIO? The reason why I used `#` as the delimiter is because a lot of our users have a whitespace character is their username and `ldgen.py` simply doesn't take it into account.

atanisoft, 2019-09-16:
Perhaps use \t (tab) instead of space or force IFS to be defined when not defined?

atanisoft, 2019-09-16:
It is also interesting that your CI fails with space whereas GitHub CI doesn't....

valeros, 2019-09-16:
So, just to recap, you're not able to compile a simple example even in the case of a clean installation of `espressif32` platform and `framework-espidf`?

atanisoft, 2019-09-16:
Correct, on windows it complains about the "#"

valeros, 2019-09-16:
Strange, but I couldn't reproduce the issue on Win10 and Win8. Could you please upload somewhere output from `pio platfom show espressif32`?

atanisoft, 2019-09-16:
I'm on Win 7 x64 (I never upgraded to Win10 and won't be anytime soon). I purged everything esp32 from my .platformio tree and triggered a new build and below is the output of your command:
```
espressif32 ~ Espressif 32
==========================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Version: f5eed10
Home: http://platformio.org/platforms/espressif32
Repository: https://github.com/platformio/platform-espressif32.git
Vendor: https://espressif.com/
License: Apache-2.0
Frameworks: arduino, espidf, pumbaa, simba

Packages
--------

Package toolchain-xtensa32
--------------------------
Type: toolchain
Requirements: ~2.50200.0
Installed: Yes
Version: 2.50200.80
Original version: 5.2.0
Description: GCC for SoC Xtensa L108

Package tool-openocd-esp32
--------------------------
Type: debugger
Requirements: ~1.1000.20181026
Installed: No (optional)

Package toolchain-esp32ulp
--------------------------
Type: toolchain
Requirements: ~1.22851.190618
Installed: No (optional)

Package framework-simba
-----------------------
Type: framework
Requirements: >=12.2.0
Installed: No (optional)

Package tool-mkspiffs
---------------------
Type: uploader
Requirements: ~2.230.0
Installed: No (optional)

Package framework-arduinoespressif32
------------------------------------
Type: framework
Requirements: ~2.10003.190916
Installed: No (optional)

Package framework-pumbaa
------------------------
Type: framework
Requirements: >=2.3.0
Installed: No (optional)

Package framework-espidf
------------------------
Type: framework
Requirements: ~3.30300.0
Installed: No (optional)

Package tool-esptoolpy
----------------------
Type: uploader
Requirements: ~1.20600.0
Installed: Yes
Version: 1.20600.0
Original version: 2.6.0
Description: ESP8266 and ESP32 serial bootloader utility

Boards
------
ID                               MCU    Frequency    Flash    RAM     Name
-------------------------------  -----  -----------  -------  ------  --------------------------------
..... (pruned)
```

And from the build output to show the frameworks since they are using GIT versions:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 #f5eed10 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 06a399b, framework-espidf 6ccb4cf, tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 34 compatible libraries
Scanning dependencies...
```

platformio.ini entries:
```
[env]
# overrides to use GIT versions rather than PIO cached copies.
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
  framework-espidf @ https://github.com/espressif/esp-idf.git#release/v3.3
platform=https://github.com/platformio/platform-espressif32.git
```

valeros, 2019-09-16:
So, according to your `platformio.ini` you're using `esp-idf` framework directly from the repository? 
```
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
  framework-espidf @ https://github.com/espressif/esp-idf.git#release/v3.3
```
That won't work, only patched version of the framework shipped with the espressif32 platform can work with the latest changes. So you need to delete that line and try again.

atanisoft, 2019-09-16:
I was using v3.2 previously and had needed the git version since PIO didn't update to pick up the newer versions with fixes.

So, what did you "fix" in ESP-IDF that forces me to only rely on the PIO copy?

valeros, 2019-09-16:
There are only 2 changes:
- embedded python modules from `requirements.txt`
- patched `kconfiglib.py` so it can recognize `#` symbol as the `ISF`.

atanisoft, 2019-09-16:
Has the kconfiglib.py update been sent to ESP-IDF? If not, please do so. In the meantime, can we switch from # to tab (\t) ?

atanisoft, 2019-09-16:
Looks like your CI doesn't like space or tab when building against GIT which doesn't make sense really since both are part of the default IFS setting.

valeros, 2019-09-16:
I'm not totally sure that `\t` will work because the logic in algorithm for parsing files entries is hardcoded to use `ifs[0]` for splitting values.

atanisoft, 2019-09-16:
[these lines](https://github.com/espressif/esp-idf/blob/release/v3.3/tools/kconfig_new/kconfiglib.py#L4118-L4119) do an inline replace of IFS entries 1+ with the first one before the split. That shouldn't be a problem. It is likely better to replace the IFS value in the env when spawning the python process.

atanisoft, 2019-09-16:
adding "--env IFS=#" works on windows 7 with the git version of IDF. The only piece remaining will be fix the pyparsing module inclusion in PIO by default. PR https://github.com/platformio/platform-espressif32/pull/227 has the IFS override

valeros, 2019-09-18:
`pyparsing` is now part of the framework package. In case you want to use framework directly from the repository please install this module in the IDE manually using `pip`.

closed 2019-09-18 by valeros