platformio/platform-espressif32 #229: 1.10.0 bug regarding abs() function

exige81, 2019-09-19:
After upgrading to the 1.10.0, I've notice a compile error in a project that had previously worked under 1.9.0.   Switching back to 1.9.0 confirms no error.

Under version 1.9.0, the following example code compiles:

```c++
uint16_t x = -100;
uint16_t y = abs(x) >> 5;
```

Under 1.10.0, the following error is generated

```c++
error: invalid operands of types '__gnu_cxx::__enable_if<true, double>::__type {aka double}' and 'int' to binary 'operator>>'
uint16_t y = abs(x) >> 5;
                       ^
```

It looks like 1.9.0 uses stdlib, while 1.10.0 uses cmath

Wondering if this is a bug or if it's intentional

valeros, 2019-09-19:
Hi @exige81! I'm not sure that this issue is specific only to PlatformIO. What framework are you using? Have you tried compiling your project in Arduino IDE?

exige81, 2019-09-19:
@valeros I'm using the arduino framework.

I just confirmed that the code compiles on the latest version of the Arduino IDE (1.8.10).

valeros, 2019-09-19:
Hm, what version of the board package is installed in Arduino IDE? I just tried v1.0.3 in Arduino IDE and it also fails with the same error.

exige81, 2019-09-19:
Oops..  had the wrong board package version installed.

Works with arduino-esp32 v1.0.2,  fails with v1.0.3.    I'll go ahead and submit the issue with espressif

closed 2019-09-19 by exige81

exige81, 2019-09-25:
For anyone else, the issue opened with espressif can be found [here](https://github.com/espressif/arduino-esp32/issues/3231)