platformio/platform-espressif32 #230: Build Issue with ESP32

lildent, 2019-09-20:
HI Guys.

Im using the below Platform.ini

```
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
```
and at the end of the build process I get this error

```
generate_section_info([".pio\build\esp32dev\ldgen.section_infos"], [".pio\build\esp32dev\libapp_update.a", ".pio\build\esp32dev\libbt.a", ".pio\build\esp32dev\libcoap.a", ".pio\build\esp32dev\libconsole.a", ".pio\build\esp32dev\libcxx.a", ".pio\build\esp32dev\libdriver.a", ".pio\build\esp32dev\libesp-tls.a", ".pio\build\esp32dev\libespcoredump.a", ".pio\build\esp32dev\libesp_adc_cal.a", ".pio\build\esp32dev\libesp_event.a", ".pio\build\esp32dev\libesp_https_ota.a", ".pio\build\esp32dev\libesp_https_server.a", ".pio\build\esp32dev\libesp_http_client.a", ".pio\build\esp32dev\libesp_http_server.a", ".pio\build\esp32dev\libesp_ringbuf.a", ".pio\build\esp32dev\libethernet.a", ".pio\build\esp32dev\libexpat.a", ".pio\build\esp32dev\libfatfs.a", ".pio\build\esp32dev\libfreemodbus.a", ".pio\build\esp32dev\libjsmn.a", ".pio\build\esp32dev\libjson.a", ".pio\build\esp32dev\liblog.a", ".pio\build\esp32dev\libmbedtls.a", ".pio\build\esp32dev\libmdns.a", ".pio\build\esp32dev\libmicro-ecc.a", ".pio\build\esp32dev\libmqtt.a", ".pio\build\esp32dev\libnewlib.a", ".pio\build\esp32dev\libnghttp.a", ".pio\build\esp32dev\libnvs_flash.a", ".pio\build\esp32dev\libopenssl.a", ".pio\build\esp32dev\libprotobuf-c.a", ".pio\build\esp32dev\libpthread.a", ".pio\build\esp32dev\libsdmmc.a", ".pio\build\esp32dev\libsmartconfig_ack.a", ".pio\build\esp32dev\libspiffs.a", ".pio\build\esp32dev\libtcpip_adapter.a", ".pio\build\esp32dev\libtcp_transport.a", ".pio\build\esp32dev\libulp.a", ".pio\build\esp32dev\libvfs.a", ".pio\build\esp32dev\libwear_levelling.a", ".pio\build\esp32dev\libxtensa-debug-module.a", ".pio\build\esp32dev\libasio.a", ".pio\build\esp32dev\libsoc.a", ".pio\build\esp32dev\libefuse.a", ".pio\build\esp32dev\libaws_iot.a", ".pio\build\esp32dev\libapp_trace.a", ".pio\build\esp32dev\libspi_flash.a", ".pio\build\esp32dev\libesp32.a", ".pio\build\esp32dev\libbootloader_support.a", ".pio\build\esp32dev\libunity.a", ".pio\build\esp32dev\liblwip.a", ".pio\build\esp32dev\libprotocomm.a", ".pio\build\esp32dev\libheap.a", ".pio\build\esp32dev\libfreertos.a", ".pio\build\esp32dev\libsodium.a", ".pio\build\esp32dev\libwpa_supplicant.a", ".pio\build\esp32dev\libwifi_provisioning.a"])
Generating .pio\build\esp32dev\libwpa_supplicant.a.sections_info
generate_project_ld_script([".pio\build\esp32dev\esp32.project.ld"], [".pio\build\esp32dev\ldgen.section_infos"])
"c:\homes\jtd\.platformio\penv\scripts\python.exe" "C:\homes\jtd\.platformio\packages\framework-espidf\tools\ldgen\ldgen.py" --sections ".pio\build\esp32dev\ldgen.section_infos" --input "C:\homes\jtd\.platformio\packages\framework-espidf\components\esp32\ld\esp32.project.ld.in" --config "c:\Users\jtd.MSPDOMAIN\Documents\PlatformIO\Projects\mspAutoClockESP32\src\sdkconfig.h" --fragments "C:\homes\jtd\.platformio\packages\framework-espidf\components\app_trace\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\esp32\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\espcoredump\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\esp_ringbuf\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\freertos\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\heap\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\lwip\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\soc\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\spi_flash\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\xtensa-debug-module\linker.lf" "C:\homes\jtd\.platformio\packages\framework-espidf\components\esp32\ld\esp32_fragments.lf" --output ".pio\build\esp32dev\esp32.project.ld" --kconfig "C:\homes\jtd\.platformio\packages\framework-espidf\Kconfig" --env "COMPONENT_KCONFIGS_PROJBUILD=C:\homes\jtd\.platformio\packages\framework-espidf\components\app_update\Kconfig.projbuild#C:\homes\jtd\.platformio\packages\framework-espidf\components\bootloader\Kconfig.projbuild#C:\homes\jtd\.platformio\packages\framework-espidf\components\esptool_py\Kconfig.projbuild#C:\homes\jtd\.platformio\packages\framework-espidf\components\partition_table\Kconfig.projbuild" --env "COMPONENT_KCONFIGS=C:\homes\jtd\.platformio\packages\framework-espidf\components\app_trace\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\aws_iot\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\bt\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\driver\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\efuse\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\esp32\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\espcoredump\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\esp_adc_cal\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\esp_event\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\esp_https_ota\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\esp_http_client\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\esp_http_server\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\ethernet\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\fatfs\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\freemodbus\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\freertos\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\heap\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\libsodium\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\log\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\lwip\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\mbedtls\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\mdns\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\mqtt\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\nvs_flash\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\openssl\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\pthread\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\spiffs\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\spi_flash\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\tcpip_adapter\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\unity\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\vfs\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\wear_levelling\Kconfig#C:\homes\jtd\.platformio\packages\framework-espidf\components\wifi_provisioning\Kconfig" --env "IDF_CMAKE=n" --env "IFS=#" --env "IDF_TARGET=\"esp32\""
Linking .pio\build\esp32dev\firmware.elf
c:/homes/jtd/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:esp32_out.ld:28: ignoring invalid character `#'
in expression
c:/homes/jtd/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:esp32_out.ld:28: syntax error
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```
Any help in fixing this ?

valeros, 2019-09-20:
Hi @lildent ! Could you please upload your project somewhere?

lildent, 2019-09-20:
I will do soon. I'm going trying creating a blank project now and see if the issue still happens

lildent, 2019-09-20:
yeah a new example works fine. I must of killed a file somewhere. going try copying my code into a new project folder. and see if that fixes it

lildent, 2019-09-20:
so after moving the files into a new project. it builds again fine.

lildent, 2019-09-20:
I did a winMerge check on the two project files. and all are the same. so im lost to why the issue was first there. 

Im happy to close this track now. as ive seem to fixed my issue

closed 2019-09-20 by lildent