platformio/platform-espressif32 #233: 1.10.0 wont compile anymore

exocode, 2019-09-27:
Hi there,

today I tried to compile my project. 

After a few hours I realized that I updated a few days ago the `Espressif 32` to version `1.10.0` (formerly I had `1.9.0`)

When I downgrade to 1.9.0 it compiles.

1.10.0 fails: 

```
> Executing task: platformio run <

Processing heltec_wifi_kit_32 (platform: espressif32; board: heltec_wifi_kit_32; framework: arduino)
--------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Mkdir("/Users/jan/Coding/PlatformIO/Projects/DEMO-ESP32-ARD-OTA-THINX/.pio/build/heltec_wifi_kit_32")
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_kit_32.html
PLATFORM: Espressif 32 1.10.0 > Heltec WiFi Kit 32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10003.190916 (1.0.3), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <THiNX32> 2.4.189
|   |-- <ArduinoJson> 5.13.1
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <EEPROM> 1.0.3
|   |-- <ESP32httpUpdate> 2.1.145
|   |   |-- <FS> 1.0
|   |   |-- <SPIFFS> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <PubSubClient(imroy)> 1.99.1
|   |   |-- <ArduinoJson> 5.13.1
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WifiManager> 0.99.9
|   |   |-- <DNSServer> 1.1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <WebServer> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
Compiling .pio/build/heltec_wifi_kit_32/src/main.cpp.o
Generating partitions .pio/build/heltec_wifi_kit_32/partitions.bin
Archiving .pio/build/heltec_wifi_kit_32/lib69a/libArduinoJson.a
Indexing .pio/build/heltec_wifi_kit_32/lib69a/libArduinoJson.a
Compiling .pio/build/heltec_wifi_kit_32/lib4c1/WiFi/ETH.cpp.o
[... 159 lines cut ...]
                 ^
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp:458:43: warning: unused variable 'previousMillis' [-Wunused-variable]
   unsigned long currentMillis = millis(), previousMillis = millis();
                                           ^
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp: In member function 'void THiNX::fetch_data()':
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp:507:17: warning: unused variable 'interval' [-Wunused-variable]
   unsigned long interval = 30000;
                 ^
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp:508:17: warning: unused variable 'currentMillis' [-Wunused-variable]
   unsigned long currentMillis = millis(), previousMillis = millis();
                 ^
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp:508:43: warning: unused variable 'previousMillis' [-Wunused-variable]
   unsigned long currentMillis = millis(), previousMillis = millis();
                                           ^
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp: In member function 'const char* THiNX::thinx_mac()':
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp:2069:51: warning: format '%X' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
   sprintf(mac_string, "5CCF%6X", ESP.getEfuseMac());
                                                   ^
lib/thinx-firmware-esp32/src/THiNX32Lib.cpp:2069:51: warning: format '%X' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
========================================= [FAILED] Took 9.96 seconds =========================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

valeros, 2019-10-03:
Hi @exocode ! Does it work in Arduino IDE? Could you also try [the upstream version of the platform](http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions) with the latest Arduino package? Thanks!

exocode, 2019-10-03:
Hi @valeros, to be honest, I've no clue, nor the Arduino IDE installed/in use (I use only VSCode). I hope someone else is familiar with it. Sorry for not able to help you

Bmooij, 2019-10-03:
It looks like there is a problem in the PubSubClient lib.
Which PubSubClient do you use?

Bmooij, 2019-10-03:
The problem is in the initialization order.

For example:
```cpp
class MyClass {
    private:
        uint8_t number1;
        uint8_t number2;
       
    public:
       MyClass(uint8_t a, uint8_t b) : number2(a), number1(b) {}
}
```
That throws an error because number2 is initialized before number1.

To fix it you can do 2 things.
Reorder the (private) fields
```cpp
class MyClass {
    private:
        uint8_t number2;
        uint8_t number1;
       
    public:
       MyClass(uint8_t a, uint8_t b) : number2(a), number1(b) {}
}
```

or change the initialization in the constructor:
```cpp
class MyClass {
    private:
        uint8_t number1;
        uint8_t number2;
       
    public:
       MyClass(uint8_t a, uint8_t b) : number1(b), number2(a) {}
}
```

Bmooij, 2019-10-03:
This a result of:
https://github.com/espressif/arduino-esp32/pull/2707

valeros, 2020-05-25:
Feel free to reopen the issue if can reproduce the issue with the latest espressif32 platform.

closed 2020-05-25 by valeros