platformio/platform-espressif32 #236: toolchain was broken on aarch64

asukiaaa, 2019-10-06:
I tried to compile program for esp32 on aarch64 environment (Ubuntu 18.04.3 64bit on Raspberry Pi 3B+) but it was failed for not found tool chain.

Error log:
```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio/build/esp32dev/src/main.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Generating partitions .pio/build/esp32dev/partitions.bin
Archiving .pio/build/esp32dev/libFrameworkArduinoVariant.a
sh: 1: xtensa-esp32-elf-ar: not found
*** [.pio/build/esp32dev/src/main.cpp.o] Error 127
*** [.pio/build/esp32dev/libFrameworkArduinoVariant.a] Error 127
Compiling .pio/build/esp32dev/FrameworkArduino/Esp.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pio/build/esp32dev/FrameworkArduino/Esp.cpp.o] Error 127
```

I found similar problem on issue thread of esphome( https://github.com/esphome/issues/issues/594 ) and the solution on the thread solved my problem.
I replaced tool chain like this.

```
mv ~/.platformio/packages/toolchain-xtensa32 /tmp
curl -sSL -o ./toolchain-xtensa32.tar.gz https://github.com/esphome/esphome-docker-base/releases/download/v1.4.0/toolchain-xtensa32.tar.gz
tar -xzvf ./toolchain-xtensa32.tar.gz -C /home/${USER}/.platformio/packages/
rm ./toolchain-xtensa32.tar.gz
```

The difference of package.json of each tool chain was number of system.

package.json of new toolchain
```
{
    "description": "GCC for SoC Xtensa L108",
    "name": "toolchain-xtensa32",
    "system": [
        "linux_aarch64"
    ],
    "url": "https://github.com/espressif/esp-idf",
    "version": "2.50200.80"
}
```

package.json of old replaced toolchain
```
{
    "description": "GCC for SoC Xtensa L108",
    "name": "toolchain-xtensa32",
    "system": [
        "linux_armv7l",
        "linux_armv8l",
        "linux_aarch64"
    ],
    "url": "https://github.com/espressif/esp-idf",
    "version": "2.50200.80"
}
```

--------------------------------

Is it possible to download tool chain only for aaruch64 on aarch64 environment to prevent the error about "not found"?

Thank you.

ivankravets, 2020-02-06:
A package for aarch64 has been deployed to our registry. 

I used [this](https://github.com/esphome/esphome-docker-base/releases/tag/v1.4.0) by @OttoWinter. Thanks!

closed 2020-02-06 by ivankravets

asukiaaa, 2020-02-06:
@ivankravets 
Thank you for updating.
I checked that I succeeded in compiling and uploading a program for esp32 on ubuntu18.04 64 bit on raspberry pi 3 (means aarch64) by removing `~/.platformio` and download again.
I also checked that I succeeded in compiling that on raspbian (32bit) on raspberry pi 3 (means maybe armv8l).

By the way I cannot find any commit to solve this problem on this repository.
Can you share the url of commit about this?
I want to know how about diff are needed.

ivankravets, 2020-02-07:
There is no commit. We just deployed a new package to our registry which supports aarch64. PlatformIO handles packages automatically based on their "package.json" manifest. You can check this package here https://bintray.com/beta/#/platformio/dl-packages/toolchain-xtensa32/2.50200.80?tab=files

asukiaaa, 2020-02-07:
I understand that there is no commit.
Thank you for updating that and sharing the information.

jpsuter, 2020-11-07:
the link for download is dead. Have you an other ?
Please make a new version to make the instal in aarch64 easier !