platformio/platform-espressif32 #237: Can't compile for c++11 on Windows 10 VsCode

assumptionsoup, 2019-10-07:
My code fails to build on windows 10 x64 on PlatformIo 4.0.3 with the following errors:
```
> Executing task in folder esp32_burn: C:\Users\jordan\.platformio\penv\Scripts\platformio.exe run <

Processing featheresp32 (platform: espressif32; board: featheresp32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 1.9.0 > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 1.0
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|-- <ESP32 BLE Arduino> 1.0.1
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\FastLED.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\bitswap.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\colorpalettes.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\colorutils.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\hsv2rgb.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\lib8tion.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\noise.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\platforms.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\power_mgt.cpp.o
Compiling .pio\build\featheresp32\libdb9\FastLED_ID126\wiring.cpp.o
Compiling .pio\build\featheresp32\src\led_manager.cpp.o
Compiling .pio\build\featheresp32\src\main.cpp.o
Compiling .pio\build\featheresp32\src\old_burn.cpp.o
Generating partitions .pio\build\featheresp32\partitions.bin
Compiling .pio\build\featheresp32\lib98d\EEPROM\EEPROM.cpp.o
Compiling .pio\build\featheresp32\lib635\SPI\SPI.cpp.o
In file included from src/led_manager.h:3:0,
                 from src\led_manager.cpp:1:
C:\Users\jordan\.platformio\lib\FastLED_ID126/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from C:\Users\jordan\.platformio\lib\FastLED_ID126/FastLED.h:65:0,
                 from src/led_manager.h:3,
                 from src\led_manager.cpp:1:
C:\Users\jordan\.platformio\lib\FastLED_ID126/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from src/led_manager.h:4:0,
                 from src\led_manager.cpp:1:
src/linalg.h:236:89: error: 'round' is not a member of 'std'
         struct std_round    { template<class A> auto operator() (A a) const -> decltype(std::round(a)) { return std::round(a); } };
                                                                                         ^
src/linalg.h:236:89: note: suggested alternative:
In file included from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal.h:34:0,
 C                from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:35,
ompiling .pio\build\featheresp32\lib84f\BLE\BLE2902.cpp.o
                 from src/led_manager.h:2,
                 from src\led_manager.cpp:1:
C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
In file included from src/led_manager.h:4:0,
                 from src\led_manager.cpp:1:
src/linalg.h:236:89: error: 'round' is not a member of 'std'
         struct std_round    { template<class A> auto operator() (A a) const -> decltype(std::round(a)) { return std::round(a); } };
                                                                                         ^
src/linalg.h:236:89: note: suggested alternative:
In file included from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:35,
                 from src/led_manager.h:2,
                 from src\led_manager.cpp:1:
C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
In file included from src/led_manager.h:4:0,
                 from src\led_manager.cpp:1:
src/linalg.h:240:103: error: 'copysign' is not a member of 'std'
         struct std_copysign { template<class A, class B> auto operator() (A a, B b) const -> decltype(std::copysign(a, b)) { return std::copysign(a, b); } };
                                                                                                       ^
src/linalg.h:240:103: note: suggested alternative:
In file included from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:35,
                 from src/led_manager.h:2,
                 from src\led_manager.cpp:1:
C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib/math.h:262:15: note:   'copysign'
Compiling .pio\build\featheresp32\lib84f\BLE\BLE2904.cpp.o
 extern double copysign _PARAMS((double, double));
               ^
In file included from src/led_manager.h:4:0,
                 from src\led_manager.cpp:1:
src/linalg.h:240:103: error: 'copysign' is not a member of 'std'
         struct std_copysign { template<class A, class B> auto operator() (A a, B b) const -> decltype(std::copysign(a, b)) { return std::copysign(a, b); } };
                                                                                                       ^
src/linalg.h:240:103: note: suggested alternative:
In file included from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:35,
                 from src/led_manager.h:2,
                 from src\led_manager.cpp:1:
C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib/math.h:262:15: note:   'copysign'
 extern double copysign _PARAMS((double, double));
               ^
Compiling .pio\build\featheresp32\lib84f\BLE\BLEAddress.cpp.o
Compiling .pio\build\featheresp32\lib84f\BLE\BLEAdvertisedDevice.cpp.o
Archiving .pio\build\featheresp32\lib635\libSPI.a
Compiling .pio\build\featheresp32\lib84f\BLE\BLEAdvertising.cpp.o
Compiling .pio\build\featheresp32\lib84f\BLE\BLEBeacon.cpp.o
Compiling .pio\build\featheresp32\lib84f\BLE\BLECharacteristic.cpp.o
Compiling .pio\build\featheresp32\lib84f\BLE\BLECharacteristicMap.cpp.o
Archiving .pio\build\featheresp32\lib98d\libEEPROM.a
Compiling .pio\build\featheresp32\lib84f\BLE\BLEClient.cpp.o
In file included from src\led_manager.cpp:3:0:
src\state.h: In member function 'void State::load()':
src\state.h:73:53: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
                     while( b = (pgm_read_byte( p++ )) )
                                                     ^
src\led_manager.cpp: In member function 'void FireFlicker::findNextUpdate()':
src\led_manager.cpp:263:23: error: 'round' is not a member of 'std'
         nextUpdate += std::round(200 + utils::randomf(100));

...

===================================================== [FAILED] Took 11.56 seconds =====================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```
I've truncated the build log because the rest is similar.

This is my platformio.ini:
```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
upload_port = /dev/ttyUSB0
build_flags = -std=c++11
monitor_speed = 115200
```
I've  also tried `build_flags = -std=gnu++11`.  Neither work.  This code compiles successfully on linux.

assumptionsoup, 2019-10-10:
I've tried to compile a simple example:

**main.cpp**
``` c++ 
#include <Arduino.h>
#include <cmath>

void setup() {
  Serial.begin(115200);
  float test = std::round(0.5);
}

void loop() {
}
```

**Results:**
```
> Executing task in folder esp32_burn: C:\Users\jordan\.platformio\penv\Scripts\platformio.exe run <

Processing featheresp32 (platform: espressif32; board: featheresp32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 1.9.0 > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 1.0
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
Compiling .pio\build\featheresp32\src\main.cpp.o
src\main.cpp: In function 'void setup()':
src\main.cpp:7:16: error: 'round' is not a member of 'std'
   float test = std::round(0.5);
                ^
src\main.cpp:7:16: note: suggested alternative:
In file included from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:35,
                 from src\main.cpp:1:
C:\Users\jordan\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
src\main.cpp:7:9: warning: unused variable 'test' [-Wunused-variable]
   float test = std::round(0.5);
         ^
*** [.pio\build\featheresp32\src\main.cpp.o] Error 1
====================================================== [FAILED] Took 3.87 seconds ======================================================
The terminal process terminated with exit code: 1
```

This looks to me like c++11 flag isn't working for some reason.  Any thoughts?

atanisoft, 2019-10-15:
@assumptionsoup the issue with std::round() looks like an issue which you should file over on https://github.com/espressif/arduino-esp32. There have been a few cases where the C++ std namespace elements were not being picked up correctly due to headers etc.

assumptionsoup, 2019-10-16:
Thanks.  I'll post this issue on espressif, but I don't believe this is an issue with std::, but rather the parts of std that are c++11 specific.  This compiles just fine:

```
#include <Arduino.h>
#include <string>

void setup() {
    std::string test = "hello world";
}

void loop() {
}
```

atanisoft, 2019-10-16:
Most of the C++11 bits should work but you are right in that there may be a few that are namespace polluted by various Arduino headers.

closed 2022-08-10 by stale[bot]