platformio/platform-espressif32 #242: COMPONENT_EMBED_TXTFILES Appends another NULL at each new compile

thomastech, 2019-10-25:
I fully understand that COMPONENT_EMBED_TXTFILES is designed to append a NULL to the embedded file at compile time. As explained here:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#embedding-binary-data

But this feature has undesirable behavior, as follows:

Per the documentation, I expect the NULL termination to be applied once at the end of the data. But each new compile adds another NULL to the source file. That is to say, compile 10 times and the original source file will now have 10 NULLs appended to it. This is not desirable. 

Besides the issue of the growing NULLs, I have to ask why does it modify the source file? There's no reason for this; It should simply add the NULL to the compiled code and leave the source file as-is. 

This would prevent the source file from growing in size with appended NULLs. And this would also prevent the source file's timestamp from changing due to the file modification by the compiler (which causes unnecessary file archiving on my daily backups).

Two questions:
1. Is there a way to force only one NULL to be added? 
2. I would prefer to use COMPONENT_EMBED_FILES since my data does not need a NULL to be added. Unfortunately this flag is currently not supported. Github Issue #220 has addressed this. Will COMPONENT_EMBED_FILES be added to the next official platformio release?

- Thomas

valeros, 2019-10-25:
Hi @thomastech ! This behavior should be fixed in the latest development branch and will be released with the next version. Would be great if you could test it with your project, just specify the following platform field:
```
platform = https://github.com/platformio/platform-espressif32.git
```

thomastech, 2019-10-25:
Thanks @valeros, the development branch solves all my reported complaints/issues. I would love to see this in the next official release. Hopefully it is released soon.
- Thomas

thomastech, 2019-10-25:
One last thing ...
My -DCOMPONENT_EMBED_FILES statement is extremely long (dozens of files) and it does not wrap in the editor. It's all on one continuous line that requires horizontal scrolling. 

It would be helpful to break up the statement into several text lines for easier reading. Ideally one source file per line. Perhaps using a line wrap character ( \) at each line break (or something similar).  

Any chance something like this is already available? If not, what's the official way to request a new feature?

- Thomas

valeros, 2019-10-28:
@thomastech Could you please pull the latest changes and try something like this:
```
[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
monitor_speed = 115200
build_flags =
  -DCONFIG_WIFI_SSID=\"ESP_AP\"
  -DCONFIG_WIFI_PASSWORD=\"MYPASS\"

board_build.embed_txtfiles = 
  src/private.pem.key
  src/certificate.pem.crt
  src/aws-root-ca.pem
```

closed 2019-10-28 by valeros

thomastech, 2019-10-28:
>  Could you please pull the latest changes and try something like this: {snip-snip}

@valeros Compiling with today's platform development release is a success. I tested _board_build.embed_txtfiles_ and _board_build.embed_files_.  Both options are working correctly. Thanks!

I'm a bit nervous using a development platform. Hopefully feedback like mine helps elevate these improvements to the next official stable version. 
 
- Thomas

commit 685e78f9 (2019-10-28): Update aws-iot example with new configuration // Resolve #242