platformio/platform-espressif32 #244: Malformed mkspiffs filepath with "framework = espidf, arduino"

boarchuz, 2019-10-30:
If "framework = espidf, arduino", then "Upload File System image" fails with:

> '"mkspiffs_espressif32_espidf arduino"' is not recognized as an internal or external command, operable program or batch file.

See:
https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L161

valeros, 2019-10-30:
Hi @boarchuz ! Looks like that the current build script is not able to handle such case. What version of `mkspiffs` tool (for arduino or esp-idf) should we use when two frameworks specified? Would it be OK if we take the first one from `platformio.ini`?

boarchuz, 2019-10-31:
Regardless of the order, Arduino will be used as a component of esp-idf, so I think esp-idf should always take precedence if it appears anywhere.
ie:
if "espidf" in $PIOFRAMEWORK
    use esp-idf (default)
else
    use $PIOFRAMEWORK

commit 1fadd964 (2019-11-01): Fix mkspiffs binary name when both arduino and esp-idf frameworks selected // Resolve #244
closed 2019-11-01 by valeros