platformio/platform-espressif32 #245: Multiple Nulls Appended to Files with COMPONENT_EMBED_TXTFILES

GrassHopper1977, 2019-10-30:
I think this is the correct place to post this. Wondering if anyone else has noticed this.

Since I last updated, I seem to be getting long chains of `NULL` characters appended to the end of files embedded with `COMPONENT_EMBED_TXTFILES`.

I've noticed before that it appends a `NULL` to the end of the file to be embedded. It hasn't been a problem before as if there was already a `NULL` it didn't apply another. However, current version seems to add a new NULL during each compile (even if the file is unchanged).

It's not a big problem, it just uses a little more memory than it should.

PlatformIO Versions
Home: 2.3.3
Core: 4.0.3
Espressif 32: V1.11.0

GrassHopper1977, 2019-10-30:
I'm using Windows 10, if that helps.

valeros, 2019-10-30:
Hi @GrassHopper1977 ! It's already fixed in dev branch. You can try something like this:
```
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev

board_build.embed_txtfiles = 
  src/private.pem.key
  src/certificate.pem.crt
  src/aws-root-ca.pem
```