platformio/platform-espressif32 #247: Project Inspect confusion

OmarAlkassab, 2019-11-16:
I compiled my ESP32 code in PlatformIO using VSCode, and get the following results:
[image]

Then I used the Project Inspect function and get the following result:
[image]

The first one shows that my code uses 14.7% of the RAM, but the second one shows 89% !
Which of them is correct?

valeros, 2019-11-20:
Hi @OmarAlkassab! Thanks for the report. Indeed, the results are different because we use different mechanisms to calculate memory utilization. ESP32 platform is quite tricky in terms of application layout and if I'm not mistaken, ESP32 has three different RAM types and some of them may contain executable data. So I'd say that the report from Project Inspector is a bit more precise. In any case, could you add a buffer of size 50kb to your application and try to build your project? Does the linker report a memory overflow?

OmarAlkassab, 2019-11-20:
I tried to add `byte buffer[50*1024] = {1};` but nothing happen, there is no difference nor in Project Inspect, nor in the compiling process!

OmarAlkassab, 2019-11-20:
I tried `byte buffer[50*1024];` and in the `loop()` function, I added `buffer[0]=1;`, the linker didn't show any overflow, but when I tried `byte buffer[500*1024];` the linker show an overflow.
[image]

OmarAlkassab, 2019-11-20:
Trying `byte buffer[60*1024];` gives the following results:
[image]

[image]

valeros, 2019-11-20:
Could you send us your project for analysis?

OmarAlkassab, 2019-11-21:
Unfortunately no, because the project is team work, and contains many confidential concepts and innovative ideas.
If you cannot help me without seeing the code, no problem, many thanks.

84berlinetta, 2020-01-24:
Does anyone have an answer for Omar's question?  I have encountered the same issue, and I am wondering which analysis of resource utilization is correct.  I have noticed that if I compile a project which exceeds the 100% indication in the inspector statistics screen, I experience issues with the run-time operation which would indicate the design is memory starved.

BTW - I am in the same position as Omar, where I cannot release the project for scrutiny.  However, I would suspect that someone could exercise one of the example ESP32 WiFi projects to investigate the differences between the resources reported by the linker vs. the inspector.

valeros, 2020-01-29:
Hi @84berlinetta,  you can find some info about this issue [here](https://community.platformio.org/t/inspect-strange-ram-size-with-esp32/11663/6?u=valeros).

84berlinetta, 2020-01-29:
Thanks valeros... this at least explains how the values are calculated, and follows my suspicions that the PlatformIO analysis is more accurate.

closed 2022-08-10 by stale[bot]