platformio/platform-espressif32 #248: Source `huge_app.csv' not found when board = esp32cam

mentarus, 2019-11-17:
Trying to compile the basic hello world with board = esp32cam and framework = espidf and getting:
`*** [.pio\build\esp32cam\partitions.bin] Source `huge_app.csv' not found, needed by target `.pio\build\esp32cam\partitions.bin'.`

Configuration issue?

Visual Studio code: 1.39.1
Windows 10.0.18632
PIO, no version information, running latest as of 17NOV19
Espressif platform 1.90

valeros, 2019-11-20:
Hi @mentarus ! Thanks for the report. Looks like `huge_app.csv` partition is only supported in Arduino framework. Could you please try to adding the following line to your `platformio.ini`
```
board_build.partitions = partitions_singleapp.csv
```

haniffm, 2020-03-16:
> Hi @mentarus ! Thanks for the report. Looks like `huge_app.csv` partition is only supported in Arduino framework. Could you please try to adding the following line to your `platformio.ini`
> 
> ```
> board_build.partitions = partitions_singleapp.csv
> ```

This solution worked for me. Thanks!

closed 2022-04-18 by valeros