platformio/platform-espressif32 #253: DS18B20 always reading 85 or -127

cajomferro, 2019-12-05:
I am trying to use the ESP32 to read temperature with the DS18B20 sensor. When I upload the example sketch from Arduino it works fine. If I upload the exact same code (from the sketch) using platformio it starts reading 85 or -127 all the time.

I know there have been some problems with the OneWire library + ESP32. For example, in the past I solved the problem by using an alternative OneWire library from [stickbreaker/OneWire](https://github.com/stickbreaker/OneWire).

What I do find weird is that now it works using Arduino IDE (even if it is the original OneWire library). Then something is different on platformio, because it doesn't work with any OneWire version.

This is the ESP32 board: https://www.ezsbc.com/index.php/featured-products-list-home-page/wifi01-35.html#.Xejjpy2tE8Y

I checked and all library versions seem to be the same in Arduino and platformio:
OneWire 2.3.5
Dallas 3.8.0
toolchain-xtensa32 2.50200.80 (5.2.0)
framework-arduinoespressif32 2.10004.191002 (1.0.4)
tool-esptoolpy 1.20600.0 (2.6.0)

What can be different?!

Thanks in advance!

Nikfinn99, 2020-04-16:
I've also had the exaxt same issue just now.
After a lot of debugging, i've found out, that for some reason the OneWire has an issue, when searching for devices the first time. If you continually search for devices, at the second search, the sensors will be found and continue to be found reliably now.
The DallasTemperature library will only perform the search

thavlik, 2020-07-18:
The above fix does not work for me. I wrote a little function to test each GPIO pin, and none of them work. Perhaps there are problems with this search algorithm. If I can't get it working after a bit more work, I'm going to shoot Daniel at EZSBC and email. I'll keep this thread updated with my findings since I am using the same board as OP.

```C++
bool try_onewire(int i)
{
  Serial.print("trying ");
  Serial.println(i);
  OneWire oneWire(i);
  DallasTemperature sensors(&oneWire);
  sensors.begin();
  sensors.begin();
  for (int j = 0; j < 10; j++)
  {
    int num_devices = sensors.getDeviceCount();
    sensors.requestTemperatures();
    float temperatureC = sensors.getTempCByIndex(0);
    if (num_devices > 0 || temperatureC > -126.99)
    {
      Serial.print(i);
      Serial.println(" works!");
      return true;
    }
    delay(500);
  }
  Serial.print(i);
  Serial.println(" doesn't work");
  return false;
}
```

thavlik, 2020-07-24:
Daniel at EZSBC offered phenomenal support on this. instructed me to follow [Rui Santos' tutorial](https://randomnerdtutorials.com/esp32-ds18b20-temperature-arduino-ide/) and use those exact versions of OneWire and DallasTemperature. He then offered the following picture and code of a working DS18B20 setup. The resistor is 5.1 kOhm and the data wire is on IO4.

[image]

```
//ESP32 DS18B20 Web server

#ifdef ESP32
  #include <WiFi.h>
  #include <ESPAsyncWebServer.h>
#else
  #include <Arduino.h>
  #include <ESP8266WiFi.h>
  #include <Hash.h>
  #include <ESPAsyncTCP.h>
  #include <ESPAsyncWebServer.h>
#endif
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is connected to GPIO 4
#define ONE_WIRE_BUS 4

// Setup a oneWire instance to communicate with any OneWire devices
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature sensor 
DallasTemperature sensors(&oneWire);

// Replace with your network credentials
const char* ssid = "ssid";
const char* password = "password";

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

String readDSTemperatureC() {
  // Call sensors.requestTemperatures() to issue a global temperature and Requests to all devices on the bus
  sensors.requestTemperatures(); 
  float tempC = sensors.getTempCByIndex(0);

  if(tempC == -127.00) {
    Serial.println("Failed to read from DS18B20 sensor");
    return "--";
  } else {
    Serial.print("Temperature Celsius: ");
    Serial.println(tempC); 
  }
  return String(tempC);
}

String readDSTemperatureF() {
  // Call sensors.requestTemperatures() to issue a global temperature and Requests to all devices on the bus
  sensors.requestTemperatures(); 
  float tempF = sensors.getTempFByIndex(0);

  if(int(tempF) == -196){
    Serial.println("Failed to read from DS18B20 sensor");
    return "--";
  } else {
    Serial.print("Temperature Fahrenheit: ");
    Serial.println(tempF);
  }
  return String(tempF);
}

const char index_html[] PROGMEM = R"rawliteral(
<!DOCTYPE HTML><html>
<head>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="stylesheet" href="https://use.fontawesome.com/releases/v5.7.2/css/all.css" integrity="sha384-fnmOCqbTlWIlj8LyTjo7mOUStjsKC4pOpQbqyi7RrhN7udi9RwhKkMHpvLbHG9Sr" crossorigin="anonymous">
  <style>
    html {
     font-family: Arial;
     display: inline-block;
     margin: 0px auto;
     text-align: center;
    }
    h2 { font-size: 3.0rem; }
    p { font-size: 3.0rem; }
    .units { font-size: 1.2rem; }
    .ds-labels{
      font-size: 1.5rem;
      vertical-align:middle;
      padding-bottom: 15px;
    }
  </style>
</head>
<body>
  <h2>ESP DS18B20 Server</h2>
  <p>
    <i class="fas fa-thermometer-half" style="color:#059e8a;"></i> 
    <span class="ds-labels">Temperature Celsius</span> 
    <span id="temperaturec">%TEMPERATUREC%</span>
    <sup class="units">&deg;C</sup>
  </p>
  <p>
    <i class="fas fa-thermometer-half" style="color:#059e8a;"></i> 
    <span class="ds-labels">Temperature Fahrenheit</span>
    <span id="temperaturef">%TEMPERATUREF%</span>
    <sup class="units">&deg;F</sup>
  </p>
</body>
<script>
setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    if (this.readyState == 4 && this.status == 200) {
      document.getElementById("temperaturec").innerHTML = this.responseText;
    }
  };
  xhttp.open("GET", "/temperaturec", true);
  xhttp.send();
}, 2000) ;
setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    if (this.readyState == 4 && this.status == 200) {
      document.getElementById("temperaturef").innerHTML = this.responseText;
    }
  };
  xhttp.open("GET", "/temperaturef", true);
  xhttp.send();
}, 2000) ;
</script>
</html>)rawliteral";

// Replaces placeholder with DHT values
String processor(const String& var){
  //Serial.println(var);
  if(var == "TEMPERATUREC"){
    return readDSTemperatureC();
  }
  else if(var == "TEMPERATUREF"){
    return readDSTemperatureF();
  }
  return String();
}

void setup(){
  // Serial port for debugging purposes
  Serial.begin(115200);
  Serial.println();
  
  // Start up the DS18B20 library
  sensors.begin();
  
  // Connect to Wi-Fi
  WiFi.begin(ssid, password);
  Serial.println("Connecting to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  
  // Print ESP Local IP Address
  Serial.println(WiFi.localIP());

  // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/html", index_html, processor);
  });
  server.on("/temperaturec", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", readDSTemperatureC().c_str());
  });
  server.on("/temperaturef", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", readDSTemperatureF().c_str());
  });
  // Start server
  server.begin();
}
 
void loop(){
  
}
```

photomoose, 2020-09-14:
> I've also had the exaxt same issue just now.
> After a lot of debugging, i've found out, that for some reason the OneWire has an issue, when searching for devices the first time. If you continually search for devices, at the second search, the sensors will be found and continue to be found reliably now.
> 
> The DallasTemperature library will only perform the search once at the beginning.
> I fixed my issue for now by just writing:
> 
> ```
> dallas_temperature.begin()
> dallas_temperature.begin()
> ```
> 
> so by searching twice, the devices are found.

I, too, have come to this conclusion. Reverting back to OneWire v2.3.2 and the problem goes away, but all later versions seem to be affected. It would appear this only happens in Platformio - I don't see this problem using Arduino IDE.

closed 2022-08-11 by stale[bot]