platformio/platform-espressif32 #257: sdkconfig.h not working

Ced2911, 2019-12-10:
Since 1.11.1 release, #define in sdkconfig.h aren't used by esp-idf.

rohansingh, 2019-12-11:
Do you have a minimal reproduction, or details on how you determined that this is specifically a v1.11.1 issue? My project uses v1.11.1 and `sdkconfig.h` values are definitely respected.

Ced2911, 2019-12-11:
For example, this sample:
https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-storage-sdcard/
replace filename in the sample with some long filename.
In sdkconfig.h unset CONFIG_FATFS_LFN_NONE

Seem related to this commit https://github.com/platformio/platform-espressif32/commit/a4c3a868ba18e2d1767ee40179d4d2aa03a4773b#diff-b7f718c943b6efb197203ed612a3f0f9

rohansingh, 2019-12-11:
It's not enough to just unset `CONFIG_FATFS_LFN_NONE`, since it is the default. You need to also set either `CONFIG_FATFS_LFN_HEAP` or `CONFIG_FATFS_LFN_STACK`, depending on the behavior that you want. See: https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/kconfig.html#config-fatfs-long-filenames.

I think the underlying issue here is #20. It just needs to be easier to manage `sdkconfig` and `sdkconfig.h` from within PlatformIO. But `sdkconfig.h` definitely continues to work as it always has.

valeros, 2020-03-18:
Hi @Ced2911 ! Starting with ESP-IDF v4.0 we use the native configuration system. Could you please try your project with the upstream version of the platform, for example:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = espidf
```
Also take a look at updated docs for ESP-IDF:
https://docs.platformio.org/en/latest/frameworks/espidf.html

Thanks!

valeros, 2020-05-25:
Feel free to reopen the issue if you still any help.

closed 2020-05-25 by valeros