platformio/platform-espressif32 #259: multiple definition of aws certs with 1.11.1

meop, 2019-12-17:
Hello, I see that 1.11.1 added new options and changes in behavior for how txt files are embedded.

Unfortunately, it broke my environment. I now get compile errors and I do not know how to resolve them.

platformio.ini has been updated to have this at the beginning, instead of using -D variables:

`
[env:esp32dev]
framework = arduino, espidf
platform = espressif32
monitor_speed = 115200
upload_protocol = esptool
board = esp32dev
board_build.embed_txtfiles =
    certs/rsa2/private.pem.key
    certs/rsa2/certificate.pem.crt
    certs/rsa2/aws-root-ca-rsa.pem
`

my own aws.c has this near the top

`
extern const uint8_t aws_root_ca_pem_start[] asm("_binary_certs_rsa2_aws_root_ca_rsa_pem_start");
extern const uint8_t aws_root_ca_pem_end[] asm("_binary_certs_rsa2_aws_root_ca_rsa_pem_end");
extern const uint8_t certificate_pem_crt_start[] asm("_binary_certs_rsa2_certificate_pem_crt_start");
extern const uint8_t certificate_pem_crt_end[] asm("_binary_certs_rsa2_certificate_pem_crt_end");
extern const uint8_t private_pem_key_start[] asm("_binary_certs_rsa2_private_pem_key_start");
extern const uint8_t private_pem_key_end[] asm("_binary_certs_rsa2_private_pem_key_end");
`

and my build ends with this:

`
Linking .pio/build/esp32dev/firmware.elf
.pio/build/esp32dev/private.pem.key.txt.o: In function `_binary_certs_rsa2_private_pem_key_start':
(.rodata.embedded+0x0): multiple definition of `_binary_certs_rsa2_private_pem_key_start'
.pio/build/esp32dev/private.pem.key.txt.o:(.rodata.embedded+0x0): first defined here
.pio/build/esp32dev/private.pem.key.txt.o: In function `_binary_certs_rsa2_private_pem_key_end':
(.rodata.embedded+0x68c): multiple definition of `_binary_certs_rsa2_private_pem_key_end'
.pio/build/esp32dev/private.pem.key.txt.o:(.rodata.embedded+0x68c): first defined here
.pio/build/esp32dev/certificate.pem.crt.txt.o: In function `_binary_certs_rsa2_certificate_pem_crt_start':
(.rodata.embedded+0x0): multiple definition of `_binary_certs_rsa2_certificate_pem_crt_start'
.pio/build/esp32dev/certificate.pem.crt.txt.o:(.rodata.embedded+0x0): first defined here
.pio/build/esp32dev/certificate.pem.crt.txt.o: In function `_binary_certs_rsa2_certificate_pem_crt_end':
(.rodata.embedded+0x4c9): multiple definition of `_binary_certs_rsa2_certificate_pem_crt_end'
.pio/build/esp32dev/certificate.pem.crt.txt.o:(.rodata.embedded+0x4c9): first defined here
.pio/build/esp32dev/aws-root-ca-rsa.pem.txt.o: In function `_binary_certs_rsa2_aws_root_ca_rsa_pem_start':
(.rodata.embedded+0x0): multiple definition of `_binary_certs_rsa2_aws_root_ca_rsa_pem_start'
.pio/build/esp32dev/aws-root-ca-rsa.pem.txt.o:(.rodata.embedded+0x0): first defined here
.pio/build/esp32dev/aws-root-ca-rsa.pem.txt.o: In function `_binary_certs_rsa2_aws_root_ca_rsa_pem_end':
(.rodata.embedded+0x4a5): multiple definition of `_binary_certs_rsa2_aws_root_ca_rsa_pem_end'
.pio/build/esp32dev/aws-root-ca-rsa.pem.txt.o:(.rodata.embedded+0x4a5): first defined here
`

I thought maybe my original certs had become corrupt, so I created new ones, same issue.

Thanks for any help.

valeros, 2019-12-19:
Hi @meop ! Could you please upload somewhere a minimal example to reproduce the issue?

meop, 2019-12-19:
Hi @valeros, thank you for replying, here is a [link](https://drive.google.com/file/d/1pDceGTIh-RG7W1gc2eZ98tkWimn6QF1Z/view?usp=sharing) to the project.

valeros, 2019-12-19:
Many thanks, fixed. Please retest with the upstream version of the platform:
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-espressif32.git
board = ...
framework = ...
```

commit 264999a7 (2019-12-19): Fix multiple symbol definition when two frameworks selected // Resolve #259
closed 2019-12-19 by valeros