platformio/platform-espressif32 #261: Problem with MQTT and deep sleep

CharlesGodwin, 2019-12-19:
I have a problem. I ported a working sketch from the Arduino IDE into PlatformIO on VS Code using Windows 10. The sketch is a MQTT client that reads the temp from a DHT22, connects to my WiFi network and sends a message to local hosted MQTT broker and then goes to deep sleep for 2 minutes.
This has worked for weeks using my Arduino IDE build.

I have the VSC project set up to point to the Arduino libraries. It compiles clean, uploads and runs. All the print statements suggests its working fine. The code is setup with error checks to report failure. Nothing bad is reported. But the message never shows up in a client have setup to subscribe to the message. They show up using the Arduino build.
[Johnny_Five2.cpp.gz](https://github.com/platformio/platform-espressif32/files/3984294/Johnny_Five2.cpp.gz)

The source files are identical, I ran a file compare.
I've attached the source. What other info or files  would someone want to look at this and tell me what I can do to make it work?

Thanks for your help.

valeros, 2019-12-20:
Hi @CharlesGodwin ! From what I understand the problem is that the MQTT client is not able to send messages after deep sleep? Does it work if you don't use deep sleep at all? Could you please upload somewhere a minimal example (including the `platformio.ini` file) to reproduce the issue?

CharlesGodwin, 2019-12-20:
@valeros I will build a stripped down example today. The problem is that it NEVER sends a message. The program seems to think it does but the broker never gets it. This is true on initial startup and after deep sleep.

CharlesGodwin, 2019-12-20:
@valeros Good news and bad news. The good news is a stripped down program that just writes one message in setup works. That is also the bad news :( I will now start building up the program to see what makes it fail.

I am leaving the issue open for a short time in case I can identify the point of failure.

Thanks for your attention.

CharlesGodwin, 2019-12-20:
I have a variation on the problem occurring  without using deep sleep. Attached  are the arduino IDE version of the source, the platformio version of the source, the platformio.ini file and a text file showing the output of a mosquitto subscribe session.

The only difference in the 2 source files is line 12 where I define separate topics for MQTT.

This is not using deep sleep but is shutting down and restarting devices in the loop which delays for 10 seconds each time.

the arduino ide generated  version published 18 consecutive messages successfully. The same code running in platformio generated code published only 12 of 19 messages. The print messages in the program, using a serial monitor, show no indication of unsuccessful publishing.

I repeated the platforimio test with VSC shutdown and monitoring using serial.tools.miniterm and got only 2 of 14.

Some tests have the initial message not being transmitted before any of the "off" code has been executed.

Help....

[issue_261.zip](https://github.com/platformio/platform-espressif32/files/3989684/issue_261.zip)

valeros, 2019-12-20:
@CharlesGodwin I'm not sure what exactly is wrong here since only some messages are delivered successfully and that's acceptable for `QoS 0`, but different behavior in comparison with Arduino IDE is indeed strange. Could you please try to tweak flash configuration, something like this:
```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
monitor_speed = 115200
board_build.f_flash = 80000000L
board_build.flash_mode = qio

; build_flags = -DMQTT_MAX_PACKET_SIZE=512
```

CharlesGodwin, 2019-12-20:
@valeros I applied your change to my ini file, did a clean (to be sure) and an upload. That seems to have solved the problem.

I agree with your point about QOS but I'm doing this work on a home network with little no no traffic and everything should be delivered.

Thanks for your help.

I can't test my original program for an hour or to but I will.

What is the significance of the two settings?

CharlesGodwin, 2019-12-21:
@valeros I applied the two board_build options to my original program that had the problem and it is fixed.

I'll leave it to you to close the issue.

Thanks for your speedy response and help.

closed 2020-03-18 by valeros