platformio/platform-espressif32 #265: Build Issue, KeyError: 'PATH'

cataliz3er, 2019-12-29:
Hey guys,
I can't seem to get past this hiccup.
Installed pio, updated to the latest version, started a new project with the following configuration:

[env:esp32-poe]
platform = espressif32
board = esp32-poe
framework = espidf

Then I copied the hello world example file from here
[https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/main/hello_world_main.c](url) 
to the src folder.
I'm getting no warnings but when I build I get the following error:
[image]
Full text version here:
[pio build issue.txt](https://github.com/platformio/platform-espressif32/files/4008692/pio.build.issue.txt)

I setup the toolchain and path as instructed here: 
[https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup-scratch.html](url)

I tried changing the board to a wemos but I get the same error.
I even tried changing the Path variable name to PATH :))

Anybody got an idea?
Thanks!

valeros, 2020-03-18:
Hi @cataliz3er ! Starting with ESP-IDF v4.0 we use the native configuration system. Could you please try your project with the upstream version of the platform, for example:
```
[env:esp32-poe]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-poe
framework = espidf
```

Thanks!

valeros, 2020-05-25:
Feel free to reopen the issue if you still need any help.

closed 2020-05-25 by valeros