platformio/platform-espressif32 #267: Bluetoothserial link error

ajaybhargav, 2019-12-31:
I am using arduino and espidf frameworks. Getting link error when using Bluetoothserial library of arduino.

```
BluetoothSerial.cpp:(.text._ZL8_stop_btv+0x12): undefined reference to `esp_spp_disconnect'
BluetoothSerial.cpp:(.text._ZL8_stop_btv+0x18): undefined reference to `esp_spp_deinit'
BluetoothSerial.cpp:(.text._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x46): undefined reference to `esp_bt_gap_set_scan_mode'
BluetoothSerial.cpp:(.text._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x5f): undefined reference to `esp_spp_start_srv'
BluetoothSerial.cpp:(.text._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x113): undefined reference to `esp_spp_connect'
BluetoothSerial.cpp:(.text._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x136): undefined reference to `esp_spp_disconnect'
BluetoothSerial.cpp:(.text._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x4c): undefined reference to `esp_bt_gap_resolve_eir_data'
BluetoothSerial.cpp:(.text._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x5f): undefined reference to `esp_bt_gap_resolve_eir_data'
BluetoothSerial.cpp:(.text._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0xbc): undefined reference to `esp_bt_gap_cancel_discovery'
BluetoothSerial.cpp:(.text._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0xc4): undefined reference to `esp_spp_start_discovery'
BluetoothSerial.cpp:(.text._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x126): undefined reference to `esp_bt_gap_pin_reply'
BluetoothSerial.cpp:(.text._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x132): undefined reference to `esp_bt_gap_ssp_confirm_reply'
```
I am not able to figure out the issue 🤦‍♂

closed 2020-01-07 by ajaybhargav

ShenRen, 2020-02-07:
Excuse me, can someone tell me how to solve this problem?
I encountered the same error.

bear24rw, 2020-12-09:
I had the same issue but fixed it with make menuconfig and enabling this options: Component config -> Bluetooth -> Bluedroid Options -> Classic Bluetooth -> SPP

bluejazzCHN, 2021-05-16:
> I had the same issue but fixed it with make menuconfig and enabling this option: Component config -> Bluetooth -> Bluedroid Options -> Classic Bluetooth -> SPP

It's exactly!

camerakang, 2023-06-24:
在platformIO下应该怎么配置呢