platformio/platform-espressif32 #269: Can't compile on FreeBSD (toolchain-xtensa32)

CyanoFresh, 2020-01-10:
Hello. I cannot compile firmware on:
FreeBSD 11.2 x64.
Platformio 4.1.0
espressif32 1.11.1

pio run:
```
(esphome) [hass@homeassistant ~/esphome/test32]$ pio run
Processing test32 (board: esp32dev; framework: arduino; platform: espressif32)
------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
Error: Could not find a version that satisfies the requirement '~2.50200.0' for your system 'freebsd_amd64'
```

CyanoFresh, 2020-01-13:
Solution is to add compiled `toolchain-xtensa32` from here: https://github.com/trombik/toolchain-xtensa32/releases - working fine with them now.
Please add them to the dl repository

trombik, 2020-09-10:
FreeBSD users should be able to build esp32 projects by following instructions in the following repositories below.

- https://github.com/trombik/platformio-freebsd-toolchain-xtensa32
- https://github.com/trombik/platformio-freebsd-tool-ninja
- https://github.com/trombik/platformio-freebsd-tool-cmake
- https://github.com/trombik/platformio-freebsd-toolchain-esp32ulp

> Please add them to the dl repository

Please don't.

maxgerhardt, 2021-09-05:
The packages are still not in the PlatformIO registry and this was asked again about in https://community.platformio.org/t/tool-manager-fails-toolchains-for-freebsd-amd64-freebsd13/23314.

trombik, 2021-09-08:
please see https://github.com/trombik/xtensa-esp32-elf/tree/devel/devel/xtensa-esp32-elf

maxgerhardt, 2022-08-01:
Issue is still valid since no package exists for FreeBSD: https://registry.platformio.org/tools/platformio/toolchain-xtensa32/compatibility