platformio/platform-espressif32 #272: Firmware binaries are missing SHA256 hash of the original ELF

rohansingh, 2020-01-15:
When building binaries from an ELF, the ESP-IDF build system adds a SHA256 hash of the original ELF at offset `0xb0`. This is vital for setting up ESP OTA correctly, since the SHA256 hash is an important method of checking whether the device needs to update its firmware, whether an update succeeded, etc.

In pratical terms, this means that [`esp_ota_get_app_description`][1] always returns an `esp_app_desc_t*` with an empty SHA256 hash, and [`esp_ota_get_app_elf_sha256`][2] returns an empty array as well.

You can see the following logs when an ESP-IDF app built with PlatformIO boots up:

```
I (742) cpu_start: Application information:
I (747) cpu_start: Project name:     firmware
I (752) cpu_start: App version:      1.0.0
I (757) cpu_start: Compile time:     Jan 11 2020 01:47:20
I (763) cpu_start: ELF file SHA256:  0000000000000000...
I (769) cpu_start: ESP-IDF:          aafab7d68
```

The `0000000000000000` should contain the SHA256 prefix of the ELF file.

[1]: https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/system/ota.html#_CPPv427esp_ota_get_app_descriptionv
[2]: https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/system/ota.html#_CPPv426esp_ota_get_app_elf_sha256Pc6size_t

closed 2020-03-04 by valeros

ajaybhargav, 2020-03-05:
@valeros adding hash to ElfToBin will also add hash to booloader binary. So I believe this might break the bootloader?

From IDF, flag is only added if its not bootloader build:

ifndef IS_BOOTLOADER_BUILD
ESPTOOL_ELF2IMAGE_OPTIONS += --elf-sha256-offset 0xb0
endif

rohansingh, 2020-03-05:
I've been running with this patch for a while and it doesn't seem to break the bootloader. I can send another patch that restricts this to the firmware binary only though.

ajaybhargav, 2020-03-06:
That will be appropriate and inline with IDF implementation.

PR #290 (2020-03-04): Add ELF file hash to images generated by elf2image
Fixes #272.
commit 2ee164a3 (2020-03-04): Add ELF file hash to images generated by elf2image (#290) // Fixes #272.