platformio/platform-espressif32 #274: ULP Support on Arduino

GarlicToum, 2020-01-22:
Hey!

So, I just learned about platformio and was working on moving my esp32 project from Arduino over to it. I had some ULP code in the Arduino (using ulptool), and noticed that there wasn't any support for ULP in Arduino. So, I hacked `arduino.py` in the frameworks folder to do some of the ULP stuff from `espidf.py` (based on https://github.com/platformio/platform-espressif32/commit/8d08246acdeb8a828bc59d213e8bf449b9de53c5#diff-85c06424f599ebef79138c4c8b4456cc) and got it working (verified with a logic analyzer).

I'm not sure how best to proceed, since the change wasn't too complicated, but if you guys are interested in looking at the code, I can put in a PR for it.

valeros, 2020-01-22:
Hi @GarlicToum ! Thanks for reporting that it works. If I'm not mistaken, Arduino framework lacks some of files required to properly generate firmware for ULP, for example the linker script `esp32.ulp.ld` or `esp32ulp_mapgen.py` script. Am I missing something?

GarlicToum, 2020-01-22:
Yeah, I believe it's missing the linker script out of the box. I'm not sure about the mapgen stuff - I'm still fairly new to esp32 development as well.

Before it was working, I did try setting my framework to `arduino, espidf` in `platformio.ini`, but that didn't get it working for me - I'm not sure if it left anything behind that "helped" get things working. In the end I only had to change the arduino builder framework. 

Here's a link to my changes: https://github.com/GarlicToum/platform-espressif32/commit/de01daa9b18f10099e6cf531364bf8a66d352722

valeros, 2020-01-23:
To be honest, I'm not sure about these changes, it seems a bit hacky to me because it relies on the assumption that `esp-idf` is already installed. `arduino, espidf` bundle should've worked, probably there was some king of misconfiguration of `esp-idf`. Maybe there is another somewhat "native" way of using ULP with Arduino framework?

GarlicToum, 2020-01-24:
Awww shucks... Looks like I just got kinda lucky with [mis]configuration. I'm now recalling that I ran into a space issue with using `arduino, espidf`.

I know duff2013 managed to get ULP with Arduino working with ulptool (https://github.com/duff2013/ulptool). It looks like using a `esp32ulp_mapgen.py` and `esp32ulp_build_recipe.py`.

I'm still new to a lot of esp32 development, let alone how the build tools work, so thanks for taking some time to respond!

dobraMorda, 2020-03-25:
I want to use ULP in Arduino. Anybody make support for it in PIO?

kfine100, 2020-05-15:
I would also like to use ULP in the Arduino framework of PlatformIO. The Pulse example works fine for me in the espidf framework, but when I try Cpp examples with "framework = arduino, espidf" the Arduino.h header is not found. Not sure how to configure or if this is even possible. Is this supposed to work?

valeros, 2020-05-15:
Hi @kfine100 ! Have you tried this example https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan ?

kfine100, 2020-05-15:
Hi valeros, Thanks much for your quick response. Your example builds with no errors and I see it has cpp. However, when I port my ulp code to this example I end up going in circles trying to correct the errors. The environment is complex, and I feel like a chimpanzee trying to solve a Rubik's cube.

What would help me a lot is a simple Blink example that accesses the ulp with a small assembly code. Using PlatformIO with arduino, espidf framework. If that environment works then building up the assembly and C++ for my project would be straightforward. I think it would be used by other too, since the ulp is essential for battery powered devices.

BTW, my hardware is an Adafruit ESP32 Feather. not sure if there will be problems with the exp32 dev and esp-wrover-kit boards used in your example.

Thanks much for any help.

valeros, 2020-05-15:
OK, could you please test the following project. It's a compilation of sources/configurations based on two projects `wifiscan` and `ulp-pulse`. I'm not sure if it even works, but it compiles at least.
[espidf-arduino-ulp.zip](https://github.com/platformio/platform-espressif32/files/4633928/espidf-arduino-ulp.zip)
If it works, I'll add this project to our examples.
Thanks!

kfine100, 2020-05-17:
Hi Valerii

You are one righteous silicon pilot! Your project compiles and loads. I cannot confirm that it works yet.

I took your project and created two simpler projects. I include them which I recommend as examples for us ulp novices.

**espidf-arduino-ulp_simple_count** is a routine from Espressif. All it does is start the ulp processor which increments a number. The count is then printed by the c++ loop function. This routine DOES NOT call the processor to enter into deep sleep. Note that I have set the board to featheresp32 which is the hardware I have.

It works fine for me, but one odd thing is that I must delay about 500 ms after Serial.begin to see the first print from setup. This seems like a long time. But after the count increases about 10 x a second, so the ulp processor is working and sharing the count with the main processor.

**esp-arduino-ulp_wakeup** is the simplest possible way I could think to test putting to sleep and then waking up the main processor. Once called, the ulp resets its wakeup clock and then wakes the main processor. I copied the code from Espressif.

This routine does not seem to work for me. I include my serial output in **2020_05_17_ulpWakeupPrint.txt**. The ulp does wakeup the main processor, and cause = ESP_SLEEP_WAKEUP_ULP. However, it seems like there is a complete reboot. Could you confirm that this is not normal and do you have an idea what the problem is?

Thanks much, Kevin

[2020_05_17_ulpWakeupPrint.txt](https://github.com/platformio/platform-espressif32/files/4639931/2020_05_17_ulpWakeupPrint.txt)
[esp-arduino-ulp_wakeup.zip](https://github.com/platformio/platform-espressif32/files/4639932/esp-arduino-ulp_wakeup.zip)
[espidf-arduino-ulp_simple_count.zip](https://github.com/platformio/platform-espressif32/files/4639933/espidf-arduino-ulp_simple_count.zip)

valeros, 2020-05-17:
Hi Kevin, I'm no expert on ESP32 platform, so I cannot confirm that any of the peculiarities you described above are normal. There might be an issue with project configuration or with project code itself. It might be a good idea to ask someone on the official forum here https://esp32.com/viewforum.php?f=23

kfine100, 2020-05-18:
Hi Valerii

I can confirm that my wakeup program is not working. This part of the serial printout

> ets Jun  8 2016 00:22:57
> 
> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:7008
> ho 0 tail 12 room 4
> load:0x40078000,len:14076
> load:0x40080400,len:4304
> entry 0x400806e8

normally occurs. This output can be supressed by adding the line

> esp_deep_sleep_disable_rom_logging();

to the c++ initialization code. The part that follows in my .txt file is a reboot because something is wrong.

I have managed to program the ulp wakeup in Arduino IDE. I would happier if I could work with PlatformIO and will continue to try to make my example work. I attach the Arduino files in case someone wants to port them to PlatformIO.

 It would make a simpler example to use as a reference.

[ulp_wakeup.zip](https://github.com/platformio/platform-espressif32/files/4643803/ulp_wakeup.zip)

Thanks, Kevin

kororos, 2021-01-17:
@kfine100 Kevin, have you ever had any luck with moving it to platformio ?

kfine100, 2021-01-17:
Hi kororos

I believe I was able to get all the ulp functions working in Arduino and PlatformIO. However, my main application was to wakeup Bluetooth and send a message, and this does not work on the ESP32. I have heard they have a newer processor that can maintain BLE while asleep, but I have not tried it.

It's all a bit hazy because I went onto other problems, but I figure at some point I will use ULP for some low power application.

Kevin

jdgarcia99, 2021-04-09:
Is any update on this? I would like to see a platformio project with the Arduino Framework using the ulp capabilities. 

I have searched many sites but there is not much information available or it is not clear. 

When I add ULP code to my project I have been sorting with many errors and I'm still blocked.

Any update that clarifies a clear and updated method up to date?

Thanks in advance

dontsovcmc, 2024-08-08:
I would like to use ULP in the Arduino framework of PlatformIO to move my project [@waterius](https://github.com/dontsovcmc/waterius) to ESP32S2.
@ivankravets, how can I help PlatformIO to do this feature? Maybe you know the right way? I'm Python and C++ engineer and have another ultimate embedded engineer. Thanks a lot!

try-except, 2024-10-09:
Hi! Any updates on this? Big fan of the idea