platformio/platform-espressif32 #275: cannot open source file "Arduino.h"

sekt1953, 2020-01-22:
What kind of issue is this?

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

I get this error after running today's update 22/1/2020
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/home/sekt/Documents/PlatformIO/Projects/ESP32_UltraSonic/src/main.cpp).
cannot open source file "Arduino.h"
cannot open source file "Wire.h"
cannot open source file "Adafruit_GFX.h"
cannot open source file "Adafruit_SSD1306.h"
cannot open source file "Sonar.h"

the code had no problim befor running platformio update today 22/1/2020

ivankravets, 2020-01-22:
Does project build without issues?

sekt1953, 2020-01-22:
Yes, but when Write Serial.print it do not make spell complement

sekt

ivankravets, 2020-01-22:
Could you run `VSCode > View > Command Palette... > C/C++: Reset IntelliSense Databse` ? Wait some time when VScode rebuilds project.

sekt1953, 2020-01-22:
I still get the same fail
sekt

ivankravets, 2020-01-22:
@valeros does it work for you on Ubuntu?

sekt1953, 2020-01-22:
I am running Ubuntu 18.04

I can Build and upload it to the processor, but the file main.cpp is stil
shown in red color, and the problem bar show the errors after build, but
Build say SUCCESS

sekt1953, 2020-01-22:
how do I list the latest update i did on platformio

Den ons. 22. jan. 2020 kl. 20.23 skrev Svenn-Erik Krengelund Thomsen <
sekt1953@gmail.com>:

valeros, 2020-01-23:
Couldn't reproduce this issue on Ubuntu 18.04.
@sekt1953 Could you please upload your project somewhere?

sekt1953, 2020-01-24:
Hej
I have bean at cancer treatment to I will try tomorrow, I am to sleappy for
now
Sek.thomsen

ivankravets, 2020-01-24:
@sekt1953 Oh, no! :( Get well!

sekt1953, 2020-01-26:
Hej
if i open a new project everything is fine but i open a project that was
created before i updated i get error
here is an older project and a printscreen

Svenn-Erik
 ESP32_UltraSonic.x
<https://drive.google.com/file/d/1oGbgQNlyr_8EkukrYEIKvWkE5_BlBpBA/view?usp=drive_web>

sekt1953, 2020-01-26:
Hej
the ESP32_UltraSonic.x is a zip file

Svenn-Erik

Den søn. 26. jan. 2020 kl. 14.18 skrev Svenn-Erik Krengelund Thomsen <
sekt1953@gmail.com>:

sekt1953, 2020-01-26:
Hej
I did Edit
intelliSenseMode to default valur for linux
saved
removed .vscode
run platormio Rebuild inteilliSense index
closed VSC and reopen and it see to work

but I have to do it for all old project

Svenn-Erik K. Thomsen

Den søn. 26. jan. 2020 kl. 14.19 skrev Svenn-Erik Krengelund Thomsen <
sekt1953@gmail.com>:

valeros, 2020-05-25:
Feel free to reopen the issue if you still need any help.

closed 2020-05-25 by valeros

btvd, 2022-08-16:
I got this problem today. Can't fix it though... So I could build, but in my case:

#include "Ardusino.h" 

brings me error. 

"#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit 
cannot open source file "Ardusino.h""

MagnusErler, 2022-11-25:
> Ardusino

> I got this problem today. Can't fix it though... So I could build, but in my case:
> 
> #include "Ardusino.h"
> 
> brings me error.
> 
> "#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit cannot open source file "Ardusino.h""

Maybe you spelt "Arduino.h" wrong?