platformio/platform-espressif32 #281: ESP32 bootloop. Not working in Platformio. Works with Arduino IDE

creatormir, 2020-02-05:
I use platformio in visual studio code.

`#include <Arduino.h>

#define LED 2
void setup() {
  pinMode(LED,OUTPUT);
}

void loop() {
  delay(500);
  digitalWrite(LED,HIGH);
  delay(500);
  digitalWrite(LED,LOW);
}
`
Code works if downloaded via arduino ide.
Loading via Platformio does not work.
Serial monitor:

> �j
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
End. All this.
I tried changing the code. To do Serial print. Or still something. Everything says my code doesn 't even work.

platformio.ini:
`[env:esp32doit-devkit-v1]
platform = espressif32
; platform_packages = framework-arduinoespressif32 @
board = esp32doit-devkit-v1
framework = arduino
debug_load_cmds = preload
monitor_rts= 0
monitor_dtr= 0
monitor_speed = 115200`
I 've tried changing these parameters differently. But I didn 't find the result.
I tested PlatformIO 1.10.0 and 1.9.3

What do I do to work?

valeros, 2020-02-05:
Hi @creatormir ! Thanks for the info. I'd recommend you to try erasing the flash memory, just open terminal window and run the following command (I don't know what OS you have so the command is for Windows):
```
python c:\Users\YOUR_USER\.platformio\packages\tool-esptoolpy\esptool.py erase_flash
```

creatormir, 2020-02-06:
@valeros  
Thanks for the answer.
I closed the project, and I opened it up and it worked.
I think the problem was my insensitivity, which took so much time and distracted you.

closed 2020-02-06 by creatormir