platformio/platform-espressif32 #282: Support for AZ-Delivery ESP-32 Dev Kit C

Tomiol80, 2020-02-07:
Can you add support for AZ-Delivery ESP-32 Dev Kit C.
https://www.az-delivery.com/products/esp-32-dev-kit-c-v4
https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Datenblatt_a3bf98d8-6a53-4d26-8f1c-c61b1c82af39.pdf?76837

valeros, 2020-02-07:
Hi @Tomiol80! What framework do you plan to use with this board? If I'm not mistaken there is no official Arduino variant for this board.

Tomiol80, 2020-02-07:
Hi @valeros! Arduino at first, Python later. I selected now esp32dev, but board has 16MB Flash and 520KB SRAM.

closed 2020-02-07 by valeros

maxgerhardt, 2020-04-22:
@Tomiol80 does the board really have 16MB flash? Looking at https://community.platformio.org/t/how-to-change-partition-table-using-vscode-and-arduino-framework/13240 the "datasheet" claims 16MB but means at maximum 16MB external flash. The board has ESP-WROOM-32 with only 4MB of internal flash according to its schematics, module pictures, and on a user's account.

Tomiol80, 2020-04-22:
@maxgerhardt Yes I read the community post. I have been mistaken by datasheet. I ran the esptool flash_id:
esptool.py v2.8
Serial port COM5
Connecting........__
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:24:6f:b8
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

commit e431a495 (2020-02-07): Add new board AZ-Delivery ESP-32 Dev Kit // Resolve #282