platformio/platform-espressif32 #285: Will the examples be updated for esp-idf V 4?

DasVoelkel, 2020-02-11:


adrianpanicek, 2020-02-12:
Or better question, what are the plans for updating to 4.0?

valeros, 2020-02-12:
Hi @DasVoelkel @adrianpanicek ! Sure thing we're going to update `esp-idf` to the latest stable version but there is not ETA since it's a major release with a lot of changes it might take more time.

valeros, 2020-03-04:
Hi @DasVoelkel @adrianpanicek, we've added initial support for ESP-IDF v4.0 in [this branch](https://github.com/platformio/platform-espressif32/tree/feature/espidf-40). To try it out just specify that branch in your `platformio.ini` file, for example:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/espidf-40
framework = espidf
board = esp32dev
monitor_speed = 115200
``` 
Keep in mind that the project structure for ESP-IDF v4 is not backward compatible with ESP-IDF v3.x since we've switched to CMake as the source of our build configuration.

DasVoelkel, 2020-03-05:
Thank you! I've just seen your commit and was already eager to try it out!

DasVoelkel, 2020-03-05:
Do I need to do something more in particular? 
Or is the linking and compiling process mostly functionally the same although it was switched to cmake?
For some reason it seems none of my header files can be found in /include and I am not sure why

valeros, 2020-03-05:
What file is complaining? Does it work if you place that header files in the ``src`` folder?

DasVoelkel, 2020-03-05:
What I am trying to do:
[image]

Just the blink example with an additional header and cpp file and a fkt called "nothing" doing exactly as expected.
This compiles fine. Even if it's in a src sub-folder.

This does not work:
Same thing, just scaled up to a LOT of files:
[image]

Not sure what to look for now

DasVoelkel, 2020-03-05:
And after regenerating all the CMake files i get this error as well, a lot.
[image]

Of course the CMakeLists is huge as well, listing every file in src.

``` 
idf_component_register(SRCS "master_gate.cpp mg_app/mg_app.cpp mg_common/mg_common.cpp mg_core/mg_core.cpp mg_deserialiser/deserialiser.cpp mg_local_storage/mg_local_storage.cpp mg_objects/automation.cpp mg_objects/data_434_pack.cpp mg_objects/module.cpp mg_objects/module_queue_entry.cpp mg_objects/scenario.cpp mg_objects/sensor.cpp mg_objects/sensor_change_entry.cpp mg_objects/switch_entry.cpp mg_tests/mg_inet_test.cpp mg_time/mg_time.cpp mg_wifi/mg_wifi.cpp mg_wifi_config/mg_wifi_config.cpp mg_wifi_tasks/mg_web_tasks.cpp mg_wifi_utils/mg_wifi_utils.cpp pending-mg/Dusk2Dawn/Dusk2Dawn.cpp pending-mg/Dusk2Dawn/Dusk2Dawn.h pending-mg/mg_logic.cpp pending-mg/mg_web.cpp")
```

DasVoelkel, 2020-03-06:
The line

> `idf_component_register(SRCS "master_gate.cpp mg_app/mg_app.cpp mg_common/mg_common.cpp mg_core/mg_core.cpp mg_deserialiser/deserialiser.cpp mg_local_storage/mg_local_storage.cpp [...]

Was generated, not written by me, regenerating it with the new version also generates it in this way, after changing it by hand to "-encapsulate every file mentioned, it still stops at files.
It says it cannot find a file,  with a path, the file is literally in, even Ctrl,clicking the name to follow the link opens the file from VSC.
Can i help with debugging this? Could any other info be usefull ?

[image]

valeros, 2020-03-06:
Could you please run `pio platform update espressif32` and delete `CMakeLists.txt` in the `src` folder?

DasVoelkel, 2020-03-06:
I did just that o.0 but ok I can try it again

valeros, 2020-03-06:
It looks like you are not using the latest version of the development branch. Could you please try the following command:
`pio platform update https://github.com/platformio/platform-espressif32.git`

DasVoelkel, 2020-03-06:
That fixed it!
Thank you ^^, since this uses the espdev board, do i have to wait until the v4 is fully released to change this or is it save to change this to the board I own? I

DasVoelkel, 2020-03-21:
Thanks for the great help @valeros I appreciate it greatly!

closed 2020-03-21 by DasVoelkel