platformio/platform-espressif32 #288: Exporting firmware for OTA update - Arduino works, PlatformIO doesn't

chaosct, 2020-02-18:
Hi there,

I'm sorry if this evident somehow, but I can't find the answer anywhere.

I am using the `<Update.h>` library to install a new firmware in the device. If I generate the `.bin` file with Arduino (sketch→Export compiled binary) it works like a charm.

However, if I generate the `.bin` with PlatformIO it doesn't work.

If I just compile it in Arduino - not "Exporting" it - it has the same problem, which makes me suspect that I have to "Export" the `.bin` file in PlatformIO, instead of compiling it, or maybe I have to add some flags to make it suitable to OTA update.

Any Ideas?
Thanks,
C

hzhh110, 2020-04-16:
I use the platformio of vscode to program directly, the module can run normally, but the firmware.bin that I directly generated cannot be programmed, and the generated .bin cannot be used.

chaosct, 2020-04-16:
For what it's worth, if I run the upload command the generated .bin is valid. If I only compile, the .bin is not valid.

hzhh110, 2020-04-17:
0x10000 Start burning!

hzhh110, 2020-04-18:
esp32s use board_build.partitions = default.csv

hzhh110, 2020-04-18:
default_16MB.csv,Now the chip is 32m, how to set it better?

roener, 2020-07-09:
@chaosct 
That sounds similar to a problem I had with an Arduino MKR GSM 1400.
Here is what I did to fix it for me. Maybe it's helpful to you, too.

For the SAMD boards there is a library called "SDU" for OTA update via SD card.
https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SDU/src

There seems to be a difference in the way how PlatformIO and Arduino IDE compile the code: In Arduino IDE it is working as expected, in PlatformIO it's not.
Here is what I did to fix it:
1. Usually you `#include <SDU.h>` in your code. Change it to `#include <SDU_fixed.h>`.
2. Make a new file "SDU_fixed.h":

```
#ifndef SDU_fixed_h
#define SDU_fixed_h

#include <Arduino.h>

// copy and paste code from h-file according to your board model from https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SDU/src/boot
// here: mkrgsm1400.h
__attribute__((section(".sketch_boot"))) unsigned char sduBoot[0x4000] = {
    0x00, 0x80, 0x00, 0x20, 0x95, 0x3b, 0x00, 0x00, 0x7d, 0x3b, 0x00, 0x00,
    0x7d, 0x3b, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
...
    0x1d, 0x48, 0x00, 0x00, 0xc1, 0x4a, 0x00, 0x00, 0xb5, 0x20, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00};

#endif // SDU_fixed_h
```

3. The difference to the original library is that we copy and paste the board specific code for sduBoot directly instead of including the file (here: mkrgsm1400.h) containing the code. Here you have to copy and paste the code according to your board model. (Then of course your code is only working for that specific model.)

First, you could try this fix with the example here https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SDU/examples/Usage
An indicator of whether it worked or not is when the file size of the firmware.bin increases by 16KB.
("firmware.bin" need to be renamed to "UPDATE.bin" before putting it on the SD card)

closed 2022-09-20 by stale[bot]