platformio/platform-espressif32 #289: Enable CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID

BlueAndi, 2020-02-24:
Please enable CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID to be able to get which task is pinned to which core in the TaskStatus_t structure of FreeRTOS.

valeros, 2020-05-25:
Hi @BlueAndi ! Sorry for the late reply. If you're using ESP-IDF you can use a special command `pio run -t menuconfig` to select required configuration. In case with Arduino, we don't have control over configuration flags set in the `sdkconfig.h`.

closed 2020-05-25 by valeros

BlueAndi, 2020-05-25:
Hi @valeros, no problem. Does platformio uses for the Arduino framework precompiled libs? Where is the source fir them on github?

valeros, 2020-05-25:
@BlueAndi We use precompiled libraries shipped with Arduino from here https://github.com/espressif/arduino-esp32/tree/master/tools/sdk. The sources for that libs can be found in the official ESP-IDF repository https://github.com/espressif/esp-idf

BlueAndi, 2020-05-25:
@valeros Thanks.