platformio/platform-espressif32 #292: EnvironmentError: cannot import name Serial

Hokoshi, 2020-02-29:
PIO Core Call Error: "The current working directory C:\\Users\\hoko\\Documents\\PlatformIO\\Projects\\200226-102409-espidf-ble-adv will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\hoko\\Documents\\PlatformIO\\Projects\\200226-102409-espidf-ble-adv\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing nano32 (platform: espressif32; framework: espidf; board: nano32)\n-------------------------------------------------------------------------------\nEnvironmentError: cannot import name Serial:\n  File \"C:\\l\\tools\\python27\\lib\\site-packages\\platformio\\builder\\main.py\", line 89:\n    env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)\n  File \"C:\\Users\\hoko\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Defaults.py\", line 88:\n    _default_env = SCons.Environment.Environment(*args, **kw)\n  File \"C:\\Users\\hoko\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 992:\n    apply_tools(self, tools, toolpath)\n  File \"C:\\Users\\hoko\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 107:\n    env.Tool(tool)\n  File \"C:\\Users\\hoko\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 1798:\n    tool = SCons.Tool.Tool(tool, toolpath, **kw)\n  File \"C:\\Users\\hoko\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Tool\\__init__.py\", line 104:\n    module = self._tool_module()\n  File \"C:\\Users\\hoko\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Tool\\__init__.py\", line 125:\n    raise SCons.Errors.EnvironmentError(e)\n========================= [FAILED] Took 4.59 seconds =========================\n\nEnvironment     Status    Duration\n--------------  --------  ------------\nnano32          FAILED    00:00:04.594\nesp wrover kit  IGNORED\nesp32dev        IGNORED\n==================== 1 failed, 0 succeeded in 00:00:04.594 ===================="

ivankravets, 2020-02-29:
Does it work if you try building project?

valeros, 2022-04-18:
Closing as no further info provided. Feel free to reopen the issue if you're still experiencing the problem with the latest upstream version of the platform.

closed 2022-04-18 by valeros