platformio/platform-espressif32 #296: liniking fails with 'nonconstant expression for origin'

Floyddotnet, 2020-03-04:
hello, i am migrating an existing project which previously used "arduino" as framework to "espidf, arduino" as framework.

therefore I created a sdkconfig and transformed it to sdkconfig.h

during linking, now I get an error message with is not clear for me how to solve this:

```
Linking .pio\build\esp32dev-ble-1\firmware.elf
esp32_out.ld:64: nonconstant expression for origin
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev-ble-1\firmware.elf] Error 1
```

the only one relevant google result to this error is: [ linker scripts do not accept symbols in MEMORY section](https://sourceware.org/bugzilla/show_bug.cgi?id=4643)

can someone maybe tell me what I can do now?

```
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - framework-espidf 3.30300.190916 (3.3.0)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
```

my main goal is to get a newer esp-idf version in this project because im running into ble/wifi cooexistence problems with ESP-IDF 3.2 from arduino framework.

this is my sdkconfig.h 

```
#define CONFIG_IDF_TARGET_ESP32 1
#define CONFIG_IDF_TARGET "esp32"
#define CONFIG_IDF_FIRMWARE_CHIP_ID 0x0000

//
// SDK tool configuration
//
#define CONFIG_SDK_TOOLPREFIX "xtensa-esp32-elf-"
#define CONFIG_APP_COMPILE_TIME_DATE 1
// CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
// CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
// CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
// CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
// CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
#define CONFIG_BOOTLOADER_LOG_LEVEL_INFO 1
// CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
// CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
#define CONFIG_BOOTLOADER_LOG_LEVEL 3
// CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
#define CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V 1
// CONFIG_BOOTLOADER_FACTORY_RESET is not set
// CONFIG_BOOTLOADER_APP_TEST is not set
#define CONFIG_BOOTLOADER_WDT_ENABLE 1
// CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
#define CONFIG_BOOTLOADER_WDT_TIME_MS 9000
// CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
// CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
// CONFIG_SECURE_BOOT_ENABLED is not set
// CONFIG_SECURE_FLASH_ENC_ENABLED is not set
#define CONFIG_ESPTOOLPY_BAUD_OTHER_VAL 115200
// CONFIG_ESPTOOLPY_FLASHMODE_QIO is not set
// CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
#define CONFIG_ESPTOOLPY_FLASHMODE_DIO 1
// CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
#define CONFIG_ESPTOOLPY_FLASHMODE "dio"
// CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
#define CONFIG_ESPTOOLPY_FLASHFREQ_40M 1
[... 1059 lines cut ...]
#define CONFIG_ARDUINO_EVENT_RUN_CORE1 1
#define CONFIG_ARDUINO_EVENT_RUNNING_CORE 1
#define CONFIG_ARDUINO_UDP_RUNNING_CORE 1

#define CONFIG_DMA_RX_BUF_NUM 10
#define CONFIG_DMA_TX_BUF_NUM 10
#define CONFIG_EMAC_CHECK_LINK_PERIOD_MS 2000
#define CONFIG_EMAC_TASK_STACK_SIZE 3072
#define CONFIG_EMAC_TASK_PRIORITY 20

#define CONFIG_AWS_IOT_SDK 0
#define CONFIG_AWS_IOT_MQTT_HOST ""
#define CONFIG_AWS_IOT_MQTT_MAX_RECONNECT_WAIT_INTERVAL 128000
#define CONFIG_AWS_IOT_MQTT_MIN_RECONNECT_WAIT_INTERVAL 1000
#define CONFIG_AWS_IOT_MQTT_NUM_SUBSCRIBE_HANDLERS 5
#define CONFIG_AWS_IOT_MQTT_PORT 8883
#define CONFIG_AWS_IOT_MQTT_RX_BUF_LEN 512
#define CONFIG_AWS_IOT_MQTT_TX_BUF_LEN 512
#define CONFIG_AWS_IOT_SHADOW_MAX_JSON_TOKEN_EXPECTED 120
#define CONFIG_AWS_IOT_SHADOW_MAX_SHADOW_TOPIC_LENGTH_WITHOUT_THINGNAME 60
#define CONFIG_AWS_IOT_SHADOW_MAX_SIMULTANEOUS_ACKS 10
#define CONFIG_AWS_IOT_SHADOW_MAX_SIMULTANEOUS_THINGNAMES 10
#define CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_THING_NAME 20
#define CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_UNIQUE_CLIENT_ID_BYTES 80
```

valeros, 2020-03-04:
Hi @Floyddotnet ! Could you please compile your project in verbose mode and upload the build log to pastebin.com?

BTW, we're in the middle of switching to ESP-IDF v4 and your project might be a good test for the new build script. You can combine two examples [espidf-arduino-blink](https://github.com/platformio/platform-espressif32/tree/feature/espidf-40/examples/espidf-arduino-blink), [espidf-aws-iot](https://github.com/platformio/platform-espressif32/tree/feature/espidf-40/examples/espidf-aws-iot) and specify a special platform in your `platformio.ini`:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/espidf-40
...
```

Floyddotnet, 2020-03-04:
sure, thanks for the fast response.

the outputs and the configuration: i hope i do it right :)

```
framework = espidf, arduino
platform = espressif32@1.11.2
board = esp32dev
```
https://gist.githubusercontent.com/Floyddotnet/81bf7d6ff80107c1c51814a3e6d25838/raw/bb06eaa20488ff9f94acd554e45f353625e897e8/gistfile1.txt

```
framework = espidf, arduino
platform = https://github.com/platformio/platform-espressif32.git#feature/espidf-40
```
https://gist.githubusercontent.com/Floyddotnet/81bf7d6ff80107c1c51814a3e6d25838/raw/79c6b6d77072068783902c21e72af3deed9aab63/gistfile2.txt

```
framework = espidf, arduino
platform = https://github.com/platformio/platform-espressif32.git#feature/espidf-40
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0 
```
https://gist.githubusercontent.com/Floyddotnet/81bf7d6ff80107c1c51814a3e6d25838/raw/79c6b6d77072068783902c21e72af3deed9aab63/gistfile3.txt

```
framework = arduino
platform = https://github.com/platformio/platform-espressif32.git#feature/espidf-40
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0 
```
https://gist.githubusercontent.com/Floyddotnet/81bf7d6ff80107c1c51814a3e6d25838/raw/2f4e8affe72715a78d2d233b49de73dea449d55f/gistfile4.txt

Floyddotnet, 2020-03-04:
after solving the ringbuf_typ_t problem from the last output (disable it, i dont need it) the program compiles and run.

Floyddotnet, 2020-03-05:
should i provide an example via github?

Floyddotnet, 2020-03-05:
configuration:

```
framework = arduino, espidf
platform = espressif32
board = esp32dev
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0 
```

returns with error in `framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723\libraries\WiFi\src\ETH.cpp`:

```
error: class 'ETHClass' does not have any field named 'eth_handle'
error: class 'ETHClass' does not have any field named 'eth_link'
error: 'ETH_LINK_DOWN' was not declared in this scope
```
https://gist.githubusercontent.com/Floyddotnet/5742d10482804caf37b8af29c40f5c12/raw/1a114763d686060695de509f964bf35a974f29dd/gistfile1.txt

Floyddotnet, 2020-03-05:
configuration:

```
framework = arduino
platform = espressif32
board = esp32dev
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0 
```

returns with error in `framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723\libraries\WiFi\src\ETH.cpp`:

```
toolchain-xtensa32@2.50200.80/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:esp32_out.ld:18: syntax
collect2.exe: error: ld returned 1 exit status
```

valeros, 2020-03-05:
Hi @Floyddotnet ! Many thanks for such comprehensive reports. Could you please upload your project somewhere (github would be great). I'd like to examine what exactly breaks the compilation process.
Thanks!

Floyddotnet, 2020-03-06:
yes i can .. at first i have detailed error information for the configuration:

```
framework = arduino
platform = espressif32
board = esp32dev
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0 
```

> 
> C:\Users\Floyd\.platformio\packages\framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723\tools\sdk\lib\libbt.a(btc_config.o):(.literal.btc_config_get_bin+0xc): undefined reference to `__locale_ctype_ptr'
> C:\Users\Floyd\.platformio\packages\framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723\tools\sdk\lib\libbt.a(btc_config.o): In function `btc_config_get_bin':
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/btc/core/btc_config.c:329: undefined reference to `__locale_ctype_ptr'
> C:\Users\Floyd\.platformio\packages\framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723\tools\sdk\lib\libbt.a(config.o):(.literal.trim+0x0): undefined reference to `__locale_ctype_ptr' 
> C:\Users\Floyd\.platformio\packages\framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723\tools\sdk\lib\libbt.a(config.o): In function `trim':
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/config.c:225: undefined reference to `__locale_ctype_ptr'
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/config.c:225: undefined reference to `__locale_ctype_ptr'
> C:\Users\Floyd\.platformio\packages\framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723\tools\sdk\lib\liblwip.a(dns.o):(.literal.dns_compare_name+0x0): more undefined references to `__locale_ctype_ptr' follow
> collect2.exe: error: ld returned 1 exit status
> *** [.pio\build\esp32dev-ble-esp-idf-4\firmware.elf] Error 1

Floyddotnet, 2020-03-06:
the source: https://github.com/Floyddotnet/OpenMQTTGateway/tree/issue-296

**Config:**

prod_env.ini has 2 configurations:

esp32dev-ble-esp-idf-4_framework_arduino ; BLE libary from libs/
esp32dev-ble-esp-idf-3_framework_arduino ; BLE libary from eos1d3/ESP32_BLE#9d6c1ed

**Project details:**

The original project is: https://github.com/1technophile/OpenMQTTGateway
It use a old but modifed version of nkolban's BLE libary with has bugs and is not ESP-IDF 4 compatible.
I place the BLE libary into the libs folder and do the needed patches for ESP-IDF 4 and fix some bugs.

**Goal:**

Get a newer ESP-IDF version as 3.2 working because i running into ble/wifi coexistence problem.

valeros, 2020-03-06:
Thanks, I'll look into, but keep in mind that this config

Floyddotnet, 2020-03-06:
> Thanks, I'll look into, but keep in mind that this config won't work

this seems logical :) thank you for clarification

Floyddotnet, 2020-03-11:
@valeros .. did you have any updates for me?

i push a small fix (to https://github.com/Floyddotnet/OpenMQTTGateway/tree/issue-296), the example should now be compiling 
- with esp-idf 3.3 if the linker error can be is solved or 
- with esp-idf 4.0 if the undefined reference to `__locale_ctype_ptr' error can be fixed.

i get the same errors on ubuntu 18.03 and windows 10.  :(

**esp-idf 3.3:**

```
framework = espidf, arduino
platform = espressif32@1.11.2
board = esp32dev

/home/floyd/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32_out.ld:18: syntax error
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev-ble-esp-idf-3_framework_arduino/firmware.elf] Error 1
```

**esp-idf 4.0:**

```
framework = arduino
platform = espressif32
board = esp32dev

error: undefined reference to `__locale_ctype_ptr'

/home/floyd/.platformio/packages/framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723/tools/sdk/lib/libbt.a(btc_config.o):(.literal.btc_config_get_bin+0xc): undefined reference to `__locale_ctype_ptr'
/home/floyd/.platformio/packages/framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723/tools/sdk/lib/libbt.a(btc_config.o): In function `btc_config_get_bin':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/btc/core/btc_config.c:329: undefined reference to `__locale_ctype_ptr'
/home/floyd/.platformio/packages/framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723/tools/sdk/lib/libbt.a(config.o):(.literal.trim+0x0): undefined reference to `__locale_ctype_ptr'
/home/floyd/.platformio/packages/framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723/tools/sdk/lib/libbt.a(config.o): In function `trim':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/config.c:225: undefined reference to `__locale_ctype_ptr'
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/config.c:225: undefined reference to `__locale_ctype_ptr'
/home/floyd/.platformio/packages/framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723/tools/sdk/lib/liblwip.a(dns.o):(.literal.dns_compare_name+0x0): more undefined references to `__locale_ctype_ptr' follow
```

valeros, 2020-03-12:
Hi @Floyddotnet ! Sorry for the late reply. Have you tried  to compile your project using Arduino IDE? I'm still not sure what is the source of this issue, our build script or the Arduino framework itself. 
> `esp32_out.ld:18: syntax error`

What exactly the linker is complaining about? What is the contents of this `esp32_out.ld`?

> error: undefined reference to `__locale_ctype_ptr'

The same thing, I'm not sure about the source of this problem, looks like that revision of Arduino framework is not ready to be used with PlatformIO as a standalone package (only as a component of ESP-IDF).

BTW, is there a reason why you're trying to compile your project using only `framework = arduino` even though in the first message here you mentioned that you're porting an existing project to `espidf, arduino`?

Floyddotnet, 2020-03-12:
> BTW, is there a reason why you're trying to compile your project using only `framework = arduino` even though in the first message here you mentioned that you're porting an existing project to `espidf, arduino`?

The project originally used this configuration:

```
framework = arduino
platform = espressif32@1.11.1
board = esp32dev
```

for this configuration, an esp-idf framework version of v3.2 (v3.2.3-14-gd3e562907) will be reported when 'esp_get_idf_version' is called.

Then, I changed the platform to `espressif32@1.11.2` but the esp-idf framework version did not changed (is still 3.2).

Somewhere i found the hint that i can use "Arduino as ESP-IDF component" and then i should then get an ESP-IDF version of 3.3. As i already have some experience with the esp-idf framework without platformio, this seemed to be a viable way, and i was sure i could solve any problems and differences that might arise between arduino-esp32 and esp-idf + arduino as a component.

So i changed my configuration to:

```
framework = espidf, arduino
platform = espressif32@1.11.2
board = esp32dev
```

This works as intended and the buildscript reports now that i use "framework-espidf 3.30300.190916 (3.3.0)" but i get the descriped syntax error.

Then i open the issue here 

-------------------------------------------------------------

regarding to your comment, that i could try the new buildscript with support for ESP-IDF 4.0, i made the necessary changes to the source code and the config, and tried to compile the project. after this failed, i tried further combinations of framework, platform and platform_packages. then, i uploaded the results here.

-------------------------------------------------------------

> Have you tried to compile your project using Arduino IDE? 

No, if you wish, i can setup arduino IDE and try it. BTW: the project use travis ci and run platformio from commandline to build all variants. so even if that works, it doesn't solve my problem. :(

> I'm still not sure what is the source of this issue, our build script or the Arduino framework itself.
> 
> > `esp32_out.ld:18: syntax error`
> 
> What exactly the linker is complaining about? What is the contents of this `esp32_out.ld`?

I'm not sure. I can check / upload the `esp32_out.ld` tonight.

> > error: undefined reference to `__locale_ctype_ptr'
> 
> The same thing, I'm not sure about the source of this problem, looks like that revision of Arduino framework is not ready to be used with PlatformIO as a standalone package (only as a component of ESP-IDF).

this configuration is one of the thinks i try, regarding to your comment, that i could try the new buildscript with support for ESP-IDF 4.0

i am currently a bit confused about the many options and variations. :)

what do you think would be the correct configuration for the ESP-IDF 4.0 test from this original config. i will try it and upload the verbose log.

```
framework = arduino
platform = espressif32@1.11.1
board = esp32dev
```

valeros, 2020-03-12:
> if you wish, i can setup arduino IDE and try it. BTW: the project use travis ci and run platformio from commandline to build all variants. so even if that works, it doesn't solve my problem. :(

Yes, plase do that, at least we'll know the origin of the problem.

> I'm not sure. I can check / upload the esp32_out.ld tonight.

Yes, please check, maybe there is something obvious that we can easily fix.

> what do you think would be the correct configuration for the ESP-IDF 4.0 test from this original config. i will try it and upload the verbose log.

The config I'm expecting is something like this:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino, espidf
board = esp32dev
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0
```
But keep in mind that you'll need to manually convert `.ino` files to proper `.cpp` (basically adding `#include <Arduino.h>` where it's needed) and specify them in the project's `CMakeLists.txt`  (for example how it's done [here](https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-arduino-wifiscan/src/CMakeLists.txt)).

Floyddotnet, 2020-03-12:
esp32_out.ld (i remove all comments): 

```
MEMORY
{
  iram0_0_seg (RX) : org = 0x40080000, len = 0x20000
  iram0_2_seg (RX) : org = 0x400D0018, len = 0x330000-0x18
  dram0_0_seg (RW) : org = 0x3FFB0000 + 0xdb5c,
                                     len = 0x2c200 - 0xdb5c
  drom0_0_seg (R) : org = 0x3F400018, len = 0x400000-0x18
  rtc_iram_seg(RWX) : org = 0x400C0000, len = 0x2000
  rtc_data_seg(RW) : org = 0x3ff80000, len = 0x2000
  rtc_slow_seg(RW) : org = 0x50000000 + 0,
                                     len = 0x1000 - 0
  extern_ram_seg(RWX) : org = 0x3F800000,
                                     len = 0x400000
}
_heap_end = 0x40000000 - 0x0;
_data_seg_org = ORIGIN(rtc_data_seg);
REGION_ALIAS("rtc_data_location", rtc_slow_seg );
```

complete file: https://gist.github.com/Floyddotnet/f83c90793c26e73262cbf82b25d34ef6

valeros, 2020-03-12:
Thanks, as it turned out, you cannot have comments in `sdkconfig.h` because this file is used in  the final linker script `esp32_out.ld` , so you need to delete all unused configs:
```
// CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
```

Floyddotnet, 2020-03-13:
> Thanks, as it turned out, you cannot have comments in `sdkconfig.h` because this file is used in the final linker script `esp32_out.ld` , so you need to delete all unused configs:
> 
> ```
> // CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
> ```

ok, i will do this.

for the future, maybe you can add something like this before executiong the linker:

```
import re
with open ('file.ld', 'r' ) as f:
    content = f.read()
content = re.sub(r"(?sm)(//.*?$|/\*.*?\*/)", r"", content)
with open ('file.ld', 'w' ) as f:
    f.write(content)
```

this removes all online and multiline comments in a givven file.

Floyddotnet, 2020-03-17:
Removing all comments solved the issue. Thank you very mutch.

Did you think about my last suggestion?

closed 2020-03-17 by Floyddotnet

valeros, 2020-03-17:
@Floyddotnet There is no need as we're about to switch to ESP-IDF v4.0 which is configurable from `menuconfig` tool.