platformio/platform-espressif32 #298: Support for Heltec Wireless Stick Lite

pmelvs, 2020-03-05:
Would it be possible to get the Heltec Wireless Stick Lite board to be supported?

It's very similar to the existing Heltec Wireless Stick but uses a ESP32-PICO instead.  I tried for awhile to make a custom board but couldn't get it to work, but I'm very new to PIO and clearly don't know enough to get it working yet.

Thanks!

https://heltec.org/project/wireless-stick-lite/

brunohorta82, 2020-03-18:
nice é have the same issue.

brunohorta82, 2020-03-18:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/variants/wireless_stick_lite/pins_arduino.h

brunohorta82, 2020-03-18:
{
  "build": {
    "arduino":{
      "ldscript": "esp32_out.ld"
    },
    "core": "esp32",
    "extra_flags": "-DARDUINO_HELTEC_WIRELESS_STICK_LITE",
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "mcu": "esp32",
    "partitions": "default.csv",
    "variant": "heltec_wireless_stick_lite"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "debug": {
    "openocd_board": "esp-wroom-32.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Heltec Wireless Stick",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 1310720,
    "maximum_size": 327680,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "http://www.heltec.cn/project/wireless-stick/?lang=en",
  "vendor": "Heltec Automation"
}

brunohorta82, 2020-03-18:
Add this inside you core_dir (https://docs.platformio.org/en/latest/projectconf/section_platformio.html#projectconf-pio-core-dir) 

https://github.com/brunohorta82/platformio_heltec_wireless_stick_lite_board

pmelvs, 2020-03-18:
Thank you, this works great!  It took me a minute to figure out which board it is, in the "heltec_wireless_stick_lite.json" file you have it named as a different board made by Heltec.  I changed that and it's perfect now.

`  "name": "Heltec Wireless Stick",`

is now:

`  "name": "Heltec Wireless Stick Lite",`

aparcar, 2020-04-22:
@brunohorta82 mind creating a PR to add the board?

aparcar, 2020-04-22:
I just created the two PR mentioning you @brunohorta82 
https://github.com/platformio/platform-espressif32/pull/329
https://github.com/espressif/arduino-esp32/pull/3925

margau, 2020-04-26:
Hi, thanks for the PR!
Does this work for you? I'm compiling with PIO 4.3.1 and the following environment, but getting linker errors:
```
[env:heltec_wireless_stick_lite]
platform = https://github.com/aparcar/platform-espressif32.git#heltec_wireless_stick_lite
board = heltec_wireless_stick_lite
upload_protocol = esptool
monitor_speed = 115200
framework = arduino
monitor_filers = esp32_exception_decoder
platform_packages =
     framework-arduinoespressif32 @ https://github.com/aparcar/arduino-esp32.git#heltec_wireless_stick_lite
```

A part from my log:
```
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:27: warning: memory region `rtc_iram_seg' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:39: warning: memory region `rtc_data_seg' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:66: warning: memory region `rtc_data_location' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:107: warning: memory region `rtc_slow_seg' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:160: warning: memory region `iram0_0_seg' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:490: warning: memory region `dram0_0_seg' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:599: warning: memory region `drom0_0_seg' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32.project.ld:620: warning: memory region `iram0_2_seg' not declared
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .dram0.data loaded at [0000000000000000,00000000000021cb] overlaps section .iram0.vectors loaded at [0000000000000000,00000000000003ff]
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .flash.rodata loaded at [0000000000000000,000000000000b5ef] overlaps section .dram0.data loaded at [0000000000000000,00000000000021cb]
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .flash.text loaded at [0000000000000000,000000000001859b] overlaps section .flash.rodata loaded at [0000000000000000,000000000000b5ef]
/home/marvin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .iram0.text loaded at [0000000000000400,000000000000ad66] overlaps section .flash.text loaded at [0000000000000000,000000000001859b]
```
from there on everything fails with ` dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: __sinit` - do you have any idea, where this comes from? Under which environment are you building Wireless Stick Lite successfully?
Thanks!

commit ddab9bbb (2021-02-24): Update Arduino core to v1.0.5

Resolve #298
Resolve #329
Resolve #344
Resolve #359
Resolve #378
Resolve #384
Resolve #389
Resolve #390
Resolve #414
Resolve #420
Resolve #436
Resolve #440
Resolve #493
closed 2021-02-24 by valeros