platformio/platform-espressif32 #302: ESP-IDF 4.0 menuconfig fails

martinjaeger, 2020-03-16:
Just had a look at the ongoing v4.0 integration and I **really** appreciate the support of Kconfig and CMakeLists.txt files from native ESP-IDF. That's awesome and I hope that we can get similar support also in Zephyr in the future.

I tried to use `pio run -t menuconfig`, but get some display issues as shown below. Also the arrow keys don't work.

[image]

`idf.py menuconfig` works fine.

Running on Linux 5.4.23-1-MANJARO, Python 3.8.1.

valeros, 2020-03-16:
Hi @martinjaeger ! What version of `platformio-core` are you using?  `menuconfig` feature requires platformio-core from development branch:
`pio upgrade --dev`

martinjaeger, 2020-03-16:
Ahh, my fault. I thought I did upgrade pio core to dev already, but seems like I didn't. Works like a charm now.

Thanks a lot for the quick response!

closed 2020-03-16 by martinjaeger

pfeerick, 2020-03-26:
Just playing with the menuconfig integration (have no idea what I'm doing in ESPIDF land 😆)... and I don't have any display rendering issues, but I do see the cursor/arrow keys not working issue on Windows 10, VSCode 1.43.2, platformio-espressif32 development version, PIO 4.3.2a1. Keyboard input otherwise works with the hotkey shortcuts, esc and enter... just the cursor/arrow keys as far as I can tell.

ChrisIdema, 2020-04-07:
Upgrading to dev version doesn't work for me:
Platformio version: Core 4.3.2a1·Home 3.1.1
VSCode Version: 1.43.2
Windows 10