platformio/platform-espressif32 #306: ESP-IDF 4.0 and Arduino

jeremypoulter, 2020-03-23:
It looks like ESP-IDF does not ([will not?](https://github.com/espressif/arduino-esp32/issues/3351)) support building with Arduino as a component. Is it possible to configure the IDF version? or are we stuck at using version 1.11.2? 

If not would it be possible to do a 1.11.3 release with the backtrack decoding?

valeros, 2020-03-23:
Hi @jeremypoulter ! There is [an example](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan) with ESP-IDF v4 and Arduino as a component (although a special branch of Arduino framework is used). Is there something wrong with it?

jeremypoulter, 2020-03-23:
Silly me, only looked at my existing project, didn't see any reference to Arduino in the generated sdkconfig and the referenced issue looked to back that up. I will indeed try the example, but certainly it doesn't just work on existing projects :(

edouardreg, 2020-04-14:
Hello,

i want to use IDF v4 and Arduino as a componant. I have tried several times the Arduino wifi scan example with the special branch of Arduino framework (just import project example).
I use vscode on windows 10.
At the first time the compilation is OK.

After I run menuconfig to disable Arduino autostart, then PlatformIO make some updates (and download the special branch in framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723) and after I have compilations error with :

``
fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/edoua/.platformio/packages/framework-espidf/tools/cmake/component.cmake:171 (add_library):
  The target name
  "___idf_framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723"
  is reserved or not valid for certain CMake features, such as generator
  expressions, and may result in undefined behavior.
Call Stack (most recent call first):
  C:/Users/edoua/.platformio/packages/framework-espidf/tools/cmake/build.cmake:175 (__component_add)
  C:/Users/edoua/.platformio/packages/framework-espidf/tools/cmake/project.cmake:146 (idf_build_component)
  C:/Users/edoua/.platformio/packages/framework-espidf/tools/cmake/project.cmake:181 (__project_component_dir)
  C:/Users/edoua/.platformio/packages/framework-espidf/tools/cmake/project.cmake:243 (__project_init)
  CMakeLists.txt:3 (project)
``

If I do a clean install it's ok again, work a little on it and then the same error reappears.

bgvsan, 2020-04-14:
> Hello,
>
> i want to use IDF v4 and Arduino as a componant. I have tried several times the Arduino wifi scan example with the special branch of Arduino framework (just import project example).
> I use [...]

I can confirm I got the same problem, I'm trying to compile with the 
framework = 	arduino	espidf 
as set in the examples with no success.

`Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /Users/Diego/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
1
/Users/Diego/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot open output file a.out: No such file or directory
collect2: error: ld returned 1 exit status

Determining if the C compiler works failed with the following output:
Change Dir: /Users/Diego/Documents/PlatformIO/Projects/espidf-arduino-wifiscan/.pio/build/esp32dev/CMakeFiles/CMakeTmp

Run Build Command(s):/Users/Diego/.platformio/packages/tool-ninja/ninja cmTC_2d637 && [1/2] Building C object CMakeFiles/cmTC_2d637.dir/testCCompiler.c.obj
[2/2] Linking C executable cmTC_2d637
FAILED: cmTC_2d637 
: && /Users/Diego/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/cmTC_2d637.dir/testCCompiler.c.obj  -o cmTC_2d637   && :
/Users/Diego/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find CMakeFiles/cmTC_2d637.dir/testCCompiler.c.obj: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
`

closed 2022-07-30 by jeremypoulter