platformio/platform-espressif32 #307: ESP IDF 4.0 + Clion

MonoAnji, 2020-03-25:
After upgrade to framework 3.4 (IDF 4.0) nothing works, I can't compile for the ESP32 anymore, I can generate the ide files (pio init --ide clion) and the result doesn't match what is described at https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure, in particular it's missing
```
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```
Then again, where does that environment variable supposedly come from? Usually platformio provided that (it still does) and it didn't have to be installed again separately.

Well, the result is that compiling only results in:
```
Reading CMake configuration...
Error: Couldn't find the main target of the project!
```
from the commandline and IDE, doesn't matter. I created a new empty project for testing and that's the same, what do I have to do to make that work? Just using `pio init --ide <something>` doesn't work anymore, maybe it shouldn't have ever been working, I don't know, but it used to be enough to create a new project. Now I'm stuck... guess I'm gonna switch to the actual IDF if I need to do all kinds of things before I'm able to compile. Btw. adding the IDF_PATH line above produces other isses, could be the right way to go but will be removed when re-initializing the project anyhow.

MonoAnji, 2020-03-26:
And it seems to be very confused about which toolchain version to use
```
PLATFORM: Espressif 32 1.12.0 > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny
, tumpa)
PACKAGES:
 - framework-espidf 3.40000.200303 (4.0.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-ninja 1.7.1
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
Error: Couldn't find the main target of the project!
================================================================= [FAILED] Took 15.06 seconds =================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:15.063
native         IGNORED
============================================================ 1 failed, 0 succeeded in 00:00:15.063 ============================================================
wsl@JRPASOLDNB:/d/RPasold/dev/clion/baff$ pio platform install espressif32
PlatformManager: Installing espressif32
espressif32 @ 1.12.0 is already installed
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
toolchain-xtensa32 @ 2.50200.80 is already installed
PackageManager: Installing tool-esptoolpy @ ~1.20600.0
tool-esptoolpy @ 1.20600.0 is already installed
Uninstalling toolchain-xtensa32 @ 2.80200.200226:       [OK]
The platform 'espressif32' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
wsl@JRPASOLDNB:/d/RPasold/dev/clion/baff$ pio run
Processing esp32 (platform: espressif32; board: wemos_d1_mini32; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-xtensa32 @ ~2.80200.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-xtensa32 @ 2.80200.200226 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 1.12.0 > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny
, tumpa)
PACKAGES:
 - framework-espidf 3.40000.200303 (4.0.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-ninja 1.7.1
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
Error: Couldn't find the main target of the project!
```

valeros, 2020-03-26:
Hi @MonoAnji ! Starting with version `4.0`, `ESP-IDF` uses a build system based on CMake. In order to provide more seamless integration we also switched to CMake as the source of project build configuration. Unfortunately, there is a conflict between `CMakeLists.txt` used by `ESP-IDF`  and `CMakeLists.txt` which we generate for `CLion`. At the moment we're working on better integration with `CLion` without this intermediate "hacky" `CMakeLists.txt`, but there is no ETA for this feature.

In a nutshell, I'd recommend you to rollback to the previous version of the `espressif32` platform, e.g.:
```ini
platform = espressif32@1.11.2
```

MonoAnji, 2020-03-26:
Thank you! :) I shall do that for now.

closed 2020-03-26 by MonoAnji

Ivorforce, 2020-06-28:
Any updates on this? I'm still getting the same issue + workaround a few months past this.

lvturner, 2020-09-29:
Any updates on this?I really want to take advantage of features in the latest version of ESP-IDF so rolling back is not really an option for me.

istepaniuk, 2020-11-30:
I am having the same issue and I can't seem to make it work. Is there a workaround other than downgrading? I would really like to develop against the latest version.

dumarjo, 2020-12-11:
Hi,
Same problem here.... Need to use some new compile option on freertos...

@Ivorforce are you able to build in clion with workaround ? I try but could find any working solution

Ivorforce, 2020-12-15:
I don't remember exactly but I'm pretty sure rolling back the espressif version was a viable workaround.

g5pw, 2020-12-19:
I think this should be reopened since ESP IDF 4.x is still not compatible with the CLion build...

DavidJRobertson, 2020-12-24:
I agree, this still needs attention.

aknabi, 2021-02-03:
Still an issue... we've actually moved to Nordic as the SoC as the increased time and costs of screwing around with the ESP32 are far greater than the increase in switching components (when you're talking about only tens of thousands of devices)... they really need to sort out their tooling... it's unusable for a product, and anyone using it for a hobby would end up switching to knitting.

LizardM4, 2021-02-09:
I also agree this requires more attention. However if you are only interested in completion and syntax highlighting there may be another solution. [CLion supports compilation databases](https://www.jetbrains.com/help/clion/compilation-database.html), and [so does Platformio](https://docs.platformio.org/en/latest/integration/compile_commands.html). So for the time being I worked around this issue as follows:

1. Clean/reinit the project (I just deleted the `$PROJ_DIR/.pio` folder because I wanted to be sure it gets a clean build)
1. Remove the CMake* files used by CLion (my CLion did not pick up the compiledb automatically)
1. Generate a `compile_commands.json` via `pio run -t compiledb` (this reinitialized everything from `platformio.ini`)
2. (optional) Take `$PROJ_DIR/.pio/build/*/compile_commands.json` and remove all json entries which refer to files which reside outside of the working tree (otherwise CLion will see those as well as part of the source, which adds unnecessary clutter)
3. Copy that into the project directory. Close the project in CLion and reopen it as a compiledb project.

If everything goes as it should, you should get in the *Build* tab something like following message:
```
Cannot determine compiler type by executable file: '/home/spak/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-as'
Compilation database project successfully imported
```
Note that **features like *Run*, *Debug* are disabled with the compiledb.** Also, **do not use the CLion extension for Platformio** because it will generate broken CMakeFiles (and in my case, doing reinit would also break the build).

I just started with Platformio little time ago so I do not know how sound is this procedure, but I hope this helps. I think it could also be easily scripted or automated.

kraxor, 2021-02-20:
> Still an issue... we've actually moved to Nordic as the SoC as the increased time and costs of screwing around with the ESP32 are far greater than the increase in switching components (when you're [...]

How exactly is Espressif responsible for the shortcomings of platform.io? ESP-IDF and CLion work very well together unless you want to use platform.io (for whatever reason).

raymond-devries, 2021-04-27:
It seems that esp idf 4.0 still cannot be used with Platform IO and Clion. It would be awesome to see a fix for this in the near future!

RoscoeTheDog, 2021-05-22:
Bumping this too. 

Clion is a great IDE with so many features and PlatformIO has the best microcontroller support. It's a shame they won't play nice together with the latest update of the framework. It has been some time now since 4.0 was released and I am surprised the build conflict issue it is not grabbing the devs attention enough. The ESP32 and other espressif microcontrollers are still incredibly popular and widely used today. Those of us who use the CLion IDE will assume that platformIO will support it because of it's popularity, and just later find out after troubleshooting that you can't use the latest version of the framework.

I get it takes time to patch and fix these things but from the response to this thread so far, it seems like the devs are ignoring this? Is there a dev branch or anything working on this at all?

rikvermeer, 2021-06-06:
Why is this issue closed? It's clearly not fixed.
Please reopen

@valeros ?

valeros, 2021-06-07:
At the moment there is no plans for this feature. You can use CLion with ESP-IDF without PlatformIO https://www.jetbrains.com/help/clion/esp-idf.html

kraxor, 2021-06-11:
@valeros
If support for ESP-IDF with CLion is not even planned, then perhaps it would be worthwhile to **mention this** rather important fact somewhere **in the docs**. None of the following pages contain any information about this ~~bug~~ lack of a feature.

https://docs.platformio.org/en/latest/platforms/espressif32.html#platform-espressif32
https://docs.platformio.org/en/latest/integration/ide/clion.html#ide-clion

Support for both ESP32 and CLion without explicitly mentioning this ~~bug~~ lack of a feature **implies** that they should work together, making people spend hours trying to fix an unfixable issue, only to land here and be disappointed.

SelaliAdobor, 2021-06-23:
@valeros hi, I just came across this issue after reading that addition to the docs and wanting to see if I could contribute to any ongoing effort

But it looks like docs are a little misleading after reading this thread, since there are no plans to support this. I understand ESP-IDF's changes are putting PIO in a tough spot with the conflict, but could the messaging be updated to thoroughly recommend against using C-Lion if using planning on using an ESP-IDF based device? 

It could save people a lot of confusion if they're under the impression this is a WIP with no guaranteed ETA, vs something not planned

MrNavaStar, 2022-11-29:
This issue needs to be reopened as far as I can tell this is still broken. Is this ever going to be addressed?

Coobik, 2022-12-28:
trying to build esp idf project in platformio, getting this error:

CMake Error at /Users/USER/.platformio/packages/framework-espidf/tools/cmake/build.cmake:454 (message):
  ERROR: Because project depends on idf (>=5.0.0) which doesn't match any
  versions, version solving failed.

Call Stack (most recent call first):
  /Users/USER/.platformio/packages/framework-espidf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:4 (project)

elmot, 2023-08-18:
Recently the new CLion was released with PlatformIO plugin completely rewritten. 
ESP32 should be working now on

A blog post:
https://blog.jetbrains.com/clion/2023/07/clion-2023-2-released/#platformio_integration

CLion documentation:
https://www.jetbrains.com/help/clion/platformio.html