platformio/platform-espressif32 #309: Is there an Arduino framework for the ESP32-S module?

hzhh110, 2020-04-02:
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.3% (used 59812 bytes from 327680 bytes)
Flash: [==========] Error: The program size (1442010 bytes) is greater than maximum allowed (1310720 bytes)

platformio.ini
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino

valeros, 2020-04-02:
Hi @hzhh110 ! It looks like the default partition table is not suitable for your project. You can try other options, for example `huge_app.csv`. More info [here](https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables).

BTW, does it work in Arduino IDE?

hzhh110, 2020-04-18:
[image]
[image]

How should I configure the partition?
Mine is esp32-s 36m flash!

hzhh110, 2020-04-18:
esp32s 用 board_build.partitions = default.csv

hzhh110, 2020-04-18:
default_16MB.csv,Now the chip is 32m, how to set it better?

davidkjackson54, 2020-09-16:
I am also now getting this after adding a few more functions to my code. 
ESP32 4 Meg of Flash.
Using PlatformIO.  Core 5.0.1 Home 3.3.0 
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 61132 bytes from 327680 bytes)
Error: The program size (1514324 bytes) is greater than maximum allowed (1310720 bytes)
Flash: [====*** [checkprogsize] Explicit exit, status 1
======]  115.5% (used 1514324 bytes from 1310720 bytes)
[image]

I would appreciate any help on how I might be able to address this - if it is possible .  I am not familiar with how to override a default size with huge_app.csv for example (as mentioned above).

DennisSchm, 2021-09-16:
Hey @davidkjackson54, I don't know if you still need help but if I am correct using huge_app.csv adds flashspace by taking space reserved for over the air programming. So if you are not using this feature you are losing nothing.

closed 2022-09-20 by stale[bot]