platformio/platform-espressif32 #314: WiFi error: [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE

Milan991, 2020-04-11:
ESP32 is able to connect to WiFi, it is sending MQTT messages successfully for some time and than after 30 minutes or less, ESP32 got disconnected from the WiFi.
My program restarts the ESP32 but error `[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE` is keep repeating and ESP32 can't connect to the WiFi.

Event multiple restarts and firmware flash doesn't help, and then suddenly ESP32 connects again.

Can someone help me with this problem? Below is my connecting sequence in code:

>   WiFi.persistent(false);
>   WiFi.begin(ssid, password);
>   WiFi.setAutoReconnect(true);
>   int timeCount = 0;
>   Serial.print("Connecting");
>   while (WiFi.status() != WL_CONNECTED)
>   {
>     if ((WiFi.status() == WL_NO_SSID_AVAIL) || (WiFi.status() == WL_CONNECT_FAILED) || 30 == timeCount)
>     { //break if no ssid available, password is incorrect or not getting connected in 30sec
>       WiFi.disconnect();
>       Serial.println(F("Device restart in 5 seconds."));
>       delay(5000);
>       ESP.restart();
>     }
>     delay(1000);
>     timeCount++;
>     Serial.print(".");
>   }
>   Serial.println();
>   Serial.print("Connected, IP address : ");
>   Serial.println(WiFi.localIP());

valeros, 2020-04-13:
Hi @Milan991 ! Does it work better if you compile your project using Arduino IDE?

Milan991, 2020-04-13:
Hi @valeros , I didn't try that. I will give feedback when I try. This error is very strange and making my device unuseful.

Milan991, 2020-04-13:
@valeros Can you tell the easiest way to convert platformIO project to Arduino IDE?

valeros, 2022-04-18:
Closing as no further info provided. Feel free to reopen the issue if you're still experiencing the problem with the latest upstream version of the platform.

closed 2022-04-18 by valeros