platformio/platform-espressif32 #315: Problem when compiling ESP32 ULP sources (framework Arduino)

fededim, 2020-04-12:
I have a problem when trying to compile on platformio this [project ](https://github.com/wardjm/esp32-ulp-i2c) for ESP32, framework Arduino, with ulp source code in assembly. From the error I receive it seems that the compiler does not understand it is an assembly file (I think it tries to compile it as C/C++ file, see attached log). I have also tried to add platform_packages = toolchain-esp32ulp to platform.ini, but the error persists. Can anybody give me some hint ? How does platform io understands which compiler (e.g. executable) to use according to source file extension ?

[compilation output.txt](https://github.com/platformio/platform-espressif32/files/4466566/compilation.output.txt)

valeros, 2020-04-13:
Hi @fededim ! Try to change asm file extension from `.s` to uppercase `.S`

fededim, 2020-04-13:
Doesn't work, you closed the issue too early, please reopen it. I attach the verbose log, it seems that it uses the "xtensa-esp32-elf-gcc -x assembler-with-cpp tool" to compile the .S file but probably it's wrong.

[verbose.txt](https://github.com/platformio/platform-espressif32/files/4468784/verbose.txt)

valeros, 2020-04-13:
Does it work if you compile it using unmodified Arduino IDE?

fededim, 2020-04-13:
On unmodified Arduino IDE it does not work, you must add tools and scripts to Arduino as reported in https://github.com/duff2013/ulptool

Guiartuzo, 2020-07-22:
Hi @fededim ! Any luck with this ? I've been facing the same problem

fededim, 2020-07-22:
No I had to switch to Arduino

maxgerhardt, 2021-04-19:
This was mentioned again in https://community.platformio.org/t/esp32-ulp-coprocessor-programming-in-arduino-framework/20821

Nagymadar, 2021-06-27:
Any progress in this? This is quite essential for battery operated projects.

Jason2866, 2021-10-31:
+1 for using ULP with Arduino

nzagorec, 2021-11-19:
+1

Shame this is not supported. Probably one of the biggest setbacks of using PIO. I use ULP in almost all of my projects.

Nagymadar, 2021-11-19:
I think no one does this, although the ULP itself is almost as powerful as an AVR. :-/

likeablob, 2022-03-21:
Hi folks, FYI I wrote a thin wrapper of ulptool for PlatformIO, though it's more like a workaround rather than solution.
https://github.com/likeablob/ulptool-pio/tree/feat-pio-integration

Nagymadar, 2022-03-21:
@likeablob A step forward finally. Thanks a lot!
I started to try it right away. I am using platformio VS code on win10. It gives a script error in esp32ulp_build_recipe.py . Any idea?

`
> Executing task in folder ULP_arduino_test: C:\Users\usr\.platformio\penv\Scripts\platformio.exe run <

Processing esp32dev (board: esp32dev; platform: espressif32; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 64 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ulptool-pio> 0.1.0+sha.7c5dd85
Building in release mode
cb([".pio\build\esp32dev\firmware.elf"], [".pio\build\esp32dev\src\ulp_counter.cpp.o"])
Running ulptool
C:\Users\usr\.platformio\penv\Scripts\python.exe D:\arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio/src/esp32ulp_build_recipe.py -ID:\arduino\ULP_arduino_test\ulp -ID:\arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio\src\include\ulptool -ID:\arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio\src\ulpcc\include -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\config -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\app_trace -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\app_update -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\asio -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bootloader_support -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bt -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\coap -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\console -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\driver -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\efuse -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp-tls -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32 -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_adc_cal -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_event -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_http_client -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_http_server -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_https_ota -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_https_server -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_ringbuf -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_websocket_client -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\espcoredump -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\ethernet -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\expat -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\fatfs -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freemodbus -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\heap -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\idf_test -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\jsmn -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\json -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\libsodium -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\log -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mbedtls -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mdns -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\micro-ecc -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mqtt -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib -IC:\Users\usr\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\nghttp -IC:\Users\usr\.platformio\ [... cut ...]
  File "C:\Users\usr\.platformio\lib\ulptool-pio\post_extra_script_ulptool.py", line 54, in run_ulptool
    raise Exception("An error returned by ulptool.")
Exception: An error returned by ulptool.
============================================================================================= [FAILED] Took 1.57 seconds =============================================================================================
The terminal process "C:\Users\usr\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
`

maxgerhardt, 2022-03-21:
><command-line>:0:15: warning: missing terminating " character

Is your username containing a space by any chance?

Nagymadar, 2022-03-21:
Thanks @maxgerhardt. Anyone tried on windows and VS code? Any tips?

Nagymadar, 2022-03-21:
Tested it with a simple username. Same error.

In the provided examples, in platformio.ini I had to change the paths of pre/post scripts due to an error:

`warning: Ignoring missing SConscript 'D:\D:\Arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio\post_extra_script_ulptool.py'
`

Changed platformio.ini from:
```
extra_scripts =
  pre:/$PROJECT_LIBDEPS_DIR/$PIOENV/ulptool-pio/pre_extra_script_ulptool.py
  post:/$PROJECT_LIBDEPS_DIR/$PIOENV/ulptool-pio/post_extra_script_ulptool.py
```

To:
```
extra_scripts =
  pre:c:\Users\admin\.platformio\lib\ulptool-pio\pre_extra_script_ulptool.py
  post:c:\Users\admin\.platformio\lib\ulptool-pio\post_extra_script_ulptool.py
```

Can this be the source of the problem?

maxgerhardt, 2022-03-21:
The library correctly references `toolchain-esp32ulp` as a package (`c:\Users\<user>\.platformio\packages\toolchain-esp32ulp`), there is no need to install anything extra. The `platform.loca.txt` is also not processed by PlatformIO and not needed.

likeablob, 2022-03-22:
@maxgerhardt  @Nagymadar  Hi, sry for the late reply, and I should have mentioned that I haven't tested it on Windows. (I'm on Ubuntu.)
As @maxgerhardt  suggested, the root cause probably is in the python script itself. From the traceback, it seems a py2-py3 compatibility thing to me.

BTW, I've just enabled GitHub Issues for the repo. https://github.com/likeablob/ulptool-pio/issues
Could you open a new issue there? @Nagymadar 

```sh
Running ulptool
C:\Users\usr\.platformio\penv\Scripts\python.exe D:\arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio/src/esp32ulp_build_recipe.py -ID:\arduino\ULP_arduino_test\ulp -
... (SNIP) ...
Traceback (most recent call last):
  File "D:\arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio\src\esp32ulp_build_recipe.py", line 562, in <module>
    main(sys.argv[1:])
  File "D:\arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio\src\esp32ulp_build_recipe.py", line 107, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "D:\arduino\ULP_arduino_test\.pio\libdeps\esp32dev\ulptool-pio\src\esp32ulp_build_recipe.py", line 151, in build_ulp
    error_string = cmd[0] + '\r' + err
TypeError: can only concatenate str (not "bytes") to str
```

closed 2022-09-20 by stale[bot]

dajtxx, 2022-12-20:
Just ran into this limitation.

Thanks to @likeablob for posting the wrapper but it is only a band-aid.

Aside from the problem other people had where files had moved and directory structures changed, we hit a problem where because the ulp build seems to be done last the generated header file does not exist during the sketch build, then the build fails and the header doesn't get generated.

The work-around is to comment out all references to the ulp generated file and variables until you have a successful build without them.