platformio/platform-espressif32 #316: Libraries not working after update to new esp-idf

haberturdeur, 2020-04-13:
Hi I am having problem with libraries in my project.
For some reason after update to new version of esp-idf (v3 to v4) the compiler can't find libraries, it doesn't matter if they are in the lib folder or in platformio.ini.

```
src\BlackBox_LEDring.hpp:5:10: fatal error: SmartLeds.h: No such file or directory

*******************************************************************
* Looking for SmartLeds.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SmartLeds.h"
* Web  > https://platformio.org/lib/search?query=header:SmartLeds.h
*
```

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf

lib_deps = 
    SmartLeds
```

FoxKeys, 2020-04-16:
Confirm.
Seems like platformio.ini settings like "lib_extra_dirs", "lib_deps" stop working completely after platform update
And, have no idea, how to attach libraries using CMakeLists...

I can no longer compile any single project...

valeros, 2020-04-16:
Could you please try to roll back to the previous version of the platform:

```
[env:esp32dev]
platform = espressif32@1.11.2
framework = espidf
board = esp32dev
```

FoxKeys, 2020-04-16:
With rollback version build was successful.
This is ok as workaround (and thank you for this!). But, not as solution, of course.

closed 2020-04-16 by valeros

valeros, 2020-04-17:
Could you please retest with the upstream version of the platform, e.g.:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
```

haberturdeur, 2020-04-17:
Seems to be working, thank you very much.

FoxKeys, 2020-04-17:
On first glance - seems working. But, can't confirm. Because build stopped on error "error: exception handling disabled, use -fexceptions to enable"
(even, exceptions are enabled in platformio.ini)
`build_flags = -fexceptions`
or
`build_flags = -DPIO_FRAMEWORK_ESP_IDF_ENABLE_EXCEPTIONS, -fexceptions`
(tryed both)

Looks like build_flags required fix too...

valeros, 2020-04-17:
@FoxKeys 
Starting wiht ESP-IDF v4.0 the configuration settings like `-DPIO_FRAMEWORK_ESP_IDF_ENABLE_EXCEPTIONS` is deprecated. Please use `menuconfig` tool instead.

commit 5e0f9e47 (2020-04-16): Add missing includes from project library for ESP-IDF // Resolve #316