platformio/platform-espressif32 #317: HTTPUpdate  abort() was called at PC 0x401b968d on core 1

hzhh110, 2020-04-14:
abort() was called at PC 0x401b968d on core 1

void myOTA(const char *upgradeUrl, const char *userId, const char *currentVersion)
{
  DebugPrintln(upgradeUrl);
  String upgradeUrlStr = (String)upgradeUrl;
  upgradeUrlStr.replace("https://", "http://");
  _myDeviceDetails.isUpgrade = true;
  strcpy(_myDeviceDetails.userId, userId);
  saveDetailsConfig(_myDeviceDetails);
  sentMqttToOtaState(OTAState_UpgradeStart, userId);
  HTTPUpdate myUpdate;
  WiFiClient updateClient;
  t_httpUpdate_return ret = myUpdate.update(updateClient,upgradeUrlStr,currentVersion);
  // myUpdate.rebootOnUpdate(false);
  DebugPrintln(ret); 
  switch (ret)
  {
  case HTTP_UPDATE_FAILED:
    sentMqttToOtaState(OTAState_UpgradeFailure, userId);
    // DebugPrintF("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
    break;
  case HTTP_UPDATE_NO_UPDATES:
    sentMqttToOtaState(OTAState_UpgradeNoUpdates, userId);
    // DebugPrintln("HTTP_UPDATE_NO_UPDATES");
    break;
  case HTTP_UPDATE_OK:
    // DebugPrintln("HTTP_UPDATE_OK");
    _myDeviceDetails.upgradeSuccess = true;
    saveDetailsConfig(_myDeviceDetails);
    sentMqttToOtaState(OTAState_UpgradeSuccessful, userId);
    ESP.restart();
    break;
  }
}

valeros, 2020-04-14:
Hi @hzhh110 ! Does it work in Arduino IDE?

hzhh110, 2020-04-15:
> Hi @hzhh110 ! Does it work in Arduino IDE?

Can't? I used the vscode platform and the chip was esp32-s

The same code, I use for eps8266 firmware upgrade and normal use. In addition, one click intelligent distribution network, esp8266 can feed back to app distribution network success, and esp32 can't feed back to app distribution network success

I used WiFi function and bluetooth function to control the SCM
burning:
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.4% (used 60132 bytes from 327680 bytes)
Flash: [=====     ]  47.6% (used 1497866 bytes from 3145728 bytes)

[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino

hzhh110, 2020-04-16:
I used the platformio of vscode to write directly and it can run normally
But i use
The .pio / build / esp32cam / firmware.bin file is written with ESP32 DOWNLOAD TOOL,
appear:
CONNECT BAUD: 115200
============
....._____....._____..... test offset: 0 0x0
case ok
_____..... Efuse CONSOLE_DEBUG_DISABLE is already burned.
Uploading stub ...
Running stub ...
Stub running ...

  is stub and send flash finish

After starting up, the serial port assistant appears: 115200 baud rate
ets Jun 8 2016 00:22:57
rst: 0x10 (RTCWDT_RTC_RESET), boot: 0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

hzhh110, 2020-04-16:
I use a MAC

hzhh110, 2020-04-16:
Is burn writing only started with 0x10000? Where can I set it?

hzhh110, 2020-04-18:
esp32s 用 board_build.partitions = default.csv

hzhh110, 2020-04-18:
default_16MB.csv,Now the chip is 32m, how to set it better?

closed 2022-08-10 by stale[bot]