platformio/platform-espressif32 #318: Unable to build setting AES hardware acceleration

javoerro, 2020-04-14:
Setting sdkconfig to enable AES hardware acceleration leads to build error using mbedtls api.

sdkconfig:
`CONFIG_MBEDTLS_HARDWARE_AES=y`

code:
`#include "mbedtls/aes.h"`
`                                        `
`void app_main(void) {`
`   mbedtls_aes_context aes;`
`    mbedtls_aes_init(&aes);`
`}`

error:

> Linking .pio\build\esp32doit-devkit-v1\firmware.elf
...platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\esp-idf\src\main.c.o:(.literal.app_main+0x0): undefined reference to 'mbedtls_aes_init'
...platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\esp-idf\src\main.c.o: in function 'app_main':
...\PlatformIO\Projects\PlatformioIsssueESP32_AES_Hard/src/main.c:5: undefined reference to 'mbedtls_aes_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32doit-devkit-v1\firmware.elf] Error 1

it is supposed that CONFIG_MBEDTLS_HARDWARE_AES defines MBEDTLS_AES_ALT in `esp_config.h` in order to change the underlying code of mbdtls api by using `aes_alt.h` insted of `aes.h` but this is not working.

valeros, 2020-04-28:
Hi @javoerro ! Sorry for the late reply. The problem is caused by a non-standard name for main target. In case with PlatformIO source files are located in a special folder with the name `src` instead of the commonly agreed name `main`. This means that dependencies for the project are not collected automatically and should be specified explicitly in the `CMakeLists.txt`, e.g.:
```
idf_component_register(SRCS "main.c" REQUIRES idf::mbedtls)
```
Another option is to use `main` folder and specify it in the `platformio.ini`, for example:
```
[platformio]
src_dir = main

[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
monitor_speed = 115200
```

closed 2020-04-28 by valeros

romocitto88, 2020-05-15:
Hi @valeros I hope you could help me also.
I'm facing with the same problem, I also tried to add the line in CMakeLists.txt:

`idf_component_register(SRCS "hdl_crypto.c" REQUIRES idf::mbedtls)`

but the log output give me error:

> Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
> ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ -------------- 
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
> PLATFORM: Espressif 32 1.12.1 > Espressif ESP32 Dev Module
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-> > ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-espidf 3.40000.200303 (4.0.0)
>  - tool-cmake 3.16.4
> - tool-esptoolpy 1.20600.0 (2.6.0)
> - tool-idf 1.0.1
>  - tool-mconf 1.4060000.20190628 (406.0.0)
>  - tool-ninja 1.9.0 
> - toolchain-esp32ulp 1.22851.190618 (2.28.51)
> - toolchain-xtensa32 2.80200.200226 (8.2.0)
>Reading CMake configuration...
>-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.22.0.windows.1") 
>-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
>-- IDF_TARGET not set, using default target: esp32
>-- The C compiler identification is GNU 8.2.0
>-- The CXX compiler identification is GNU 8.2.0
>-- The ASM compiler identification is GNU
>-- Found assembler: C:/Users/ROMO/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32->elf-gcc.exe
>-- Check for working C compiler: C:/Users/ROMO/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
>-- Check for working C compiler: C:/Users/ROMO/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works
>-- Detecting C compiler ABI info
>-- Detecting C compiler ABI info - done
>-- Detecting C compile features
>-- Detecting C compile features - done
>-- Check for working CXX compiler: C:/Users/ROMO/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe
>-- Check for working CXX compiler: C:/Users/ROMO/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works
>-- Detecting CXX compiler ABI info
>-- Detecting CXX compiler ABI info - done
>-- Detecting CXX compile features
>-- Detecting CXX compile features - done
>-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
>-- Project version: 1
>-- Building ESP-IDF components for target esp32
>-- Checking Python dependencies...
>Python requirements from C:\Users\ROMO\.platformio\packages\framework-espidf\requirements.txt are satisfied.
>-- Found PythonInterp: C:/Users/ROMO/.platformio/penv/Scripts/python.exe (found version "2.7.14")
>-- Could NOT find Perl (missing: PERL_EXECUTABLE)
>-- Adding linker script C:/Users/ROMO/Documents/PlatformIO/Projects/MYPRJESP32/.pio/build/esp>32dev/esp-idf/esp32/esp32_out.ld
>-- Adding linker script C:/Users/ROMO/.platformio/packages/framework-espidf/components/esp32/ld/esp32.project.ld.in
>-- Adding linker script C:/Users/ROMO/.platformio/packages/framework-espidf/components/esp32/ld/esp32.peripherals.ld
>-- Adding linker script C:/Users/ROMO/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.ld
>-- Adding linker script C:/Users/ROMO/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
>-- Adding linker script C:/Users/ROMO/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
>-- Adding linker script C:/Users/ROMO/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
>-- Adding linker script C:/Users/ROMO/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
>-- Components: __pio_env app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs src tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
>-- Component paths: C:/Users/ROMO/Documents/PlatformIO/Projects/MyPrjEsp32/.pio/build/esp32dev/__pio_env C:/Users/ROMO/.platformio/packages/framework-espidf/components/app_trace C:/Users/ROMO/.platformio/packages/framework-espidf/components/app_update C:/Users/ROMO/.platformio/packages/framework-espidf/components/asio C:/Users/ROMO/.platformio/packages/framework-espidf/components/bootloader C:/Users/ROMO/.platformio/packages/framework-espidf/components/bootloader_support C:/Users/ROMO/.platformio/packages/framework-espidf/components/bt C:/Users/ROMO/.platformio/packages/framework-espidf/components/coap C:/Users/ROMO/.platformio/packages/framework-espidf/components/console C:/Users/ROMO/.platformio/packages/framework-espidf/components/cxx C:/Users/ROMO/.platformio/packages/framework-espidf/components/driver C:/Users/ROMO/.platformio/ [... cut ...]
fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/ROMO/.platformio/packages/framework-espidf/tools/cmake/component.cmake:445 (add_library):
  add_library cannot create target "__idf_src" because another target with
  the same name already exists.  The existing target is a static library
  created in source directory
  "C:/Users/ROMO/Documents/PlatformIO/Projects/MYPRJESP32/src".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  src/CMakeLists.txt:7 (idf_component_register)

valeros, 2020-05-15:
Hi @romocitto88 ! That's strange
- Does it work if you don't add `idf_component_register(SRCS "hdl_crypto.c" REQUIRES idf::mbedtls)`
- Could you create a new folder `main` and move your project files there, delete the old folder `src` and specify the following line in your `platformio.ini`:
```
[platformio]
src_dir = main
```
Note: if you use the `main` folder, dependencies will be resolved automatically, so there is no need in `idf_component_register(SRCS "hdl_crypto.c" REQUIRES idf::mbedtls)`.

romocitto88, 2020-05-15:
Sorry I deleted the comment to reopen the issue, I will close it.
In my CMakeList.txt file I have the following line 

`idf_component_register(SRCS ${app_sources})`

after I have added the line that you have suggest with mine:

`idf_component_register(SRCS ${app_sources})`
`idf_component_register(SRCS "hdl_crypro.c" REQUIRES idf::mbedtls)`

because I thought that only the file hdl_crypto.c required the aes. But it gave me error in the log file.
So I rewrote the line 

`idf_component_register(SRCS ${app_sources} REQUIRES idf::mbedtls)`

Now it work.