platformio/platform-espressif32 #319: Unable to compile example with framework = arduino, espidf

bgvsan, 2020-04-14:
Hi all, i've just spotted today im not able to run any example with the arduino and espidf framework set, i can compile with no problem if i remove the espidf option.

[image]

`Processing esp32dev (platform: espressif32; framework: arduino, espidf; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3746ef9
 - framework-espidf 3.40000.200303 (4.0.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\Users\diego.sa\.platformio\packages\framework-espidf\requirements.txt are satisfied.

-- Configuring incomplete, errors occurred!
See also "C:/Users/diego.sa/Documents/platform-espressif32-develop/examples/espidf-arduino-blink/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/component.cmake:171 (add_library):
  The target name
  "___idf_framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723"
  is reserved or not valid for certain CMake features, such as generator
  expressions, and may result in undefined behavior.
Call Stack (most recent call first):
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/build.cmake:175 (__component_add)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:146 (idf_build_component)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:181 (__project_component_dir)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:243 (__project_init)
  CMakeLists.txt:3 (project)

CMake Error at C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/component.cmake:19 (set_property):
  set_property could not find TARGET
  ___idf_framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723.
  Perhaps it has not yet been created.
Call Stack (most recent call first):
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/component.cmake:185 (__component_set_property)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/build.cmake:175 (__component_add)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:146 (idf_build_component)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:181 (__project_component_dir)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:243 (__project_init)
  CMakeLists.txt:3 (project)

CMake Error at C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/component.cmake:5 (get_property):
  get_property could not find TARGET
  ___idf_framework-arduinoespressif32@src-c69fc9322995db5694e2741d307d9723.
  Perhaps it has not yet been created.
Call Stack (most recent call first):
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/component.cmake:23 (__component_get_property)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/component.cmake:185 (__component_set_property)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/build.cmake:175 (__component_add)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:146 (idf_build_component)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:181 (__project_component_dir)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:243 (__project_init)
  CMakeLists.txt:3 (project)
[... 516 lines cut ...]
  CMakeLists.txt:3 (project)

CMake Error at C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/component.cmake:222 (message):
  CMake Error at
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/scripts/component_get_requirements.cmake:87
  (list):

    list sub-command REMOVE_ITEM requires two or more arguments.

Call Stack (most recent call first):
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/build.cmake:404 (__component_get_requirements)
  C:/Users/diego.sa/.platformio/packages/framework-espidf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:3 (project)

======================================================================================== [FAILED] Took 2.57 seconds ========================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:02.304
espea32        FAILED    00:00:02.933
esp320         FAILED    00:00:02.571
================================================================================== 3 failed, 0 succeeded in 00:00:07.809 ================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.`

bgvsan, 2020-04-15:
i was able to compile today changing my ini configuration to this.

[image]
 i've found this conversation from gitter that helped me.

[image]

closed 2020-04-16 by valeros

valeros, 2020-04-17:
Could you please retest with the upstream version of the platform, e.g.:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
```

pinkit-cwa, 2020-08-20:
I have a similar issue with platformio and espressif32. I get just this message, but no related messages above:
>   Some Python dependencies must be installed.  Check above message for
>   details.

@valeros 
platform = https://github.com/platformio/platform-espressif32.git does not help.

The issue in my case is that there is no executable called 'python' installed. Ubunto 20.04. I need to additionally

> $ python
> Command 'python' not found, did you mean:
>   command 'python3' from deb python3
>   command 'python' from deb python-is-python3

fixed by:

$ apt-get install python-is-python3

commit 707ae550 (2020-04-16): Fix possible issue with Arduino component for ESP-IDF // Resolve #319

CMake fails to generate code model if this path contains @ symbol