platformio/platform-espressif32 #327: New Board Request - "KEYESTUDIO Mini ESP32 Development Board with ESP-WROOM-32 Microcontroller"

jnr2820, 2020-04-21:
Hi All,

I'd like to request to get this board added. I recently purchased it and I'm not seeing it in the board list.

Baord Title from wiki: 
"KS0413 keyestudio ESP32 Core Board"

Manufacturer's wiki page: 
https://wiki.keyestudio.com/KS0413_keyestudio_ESP32_Core_Board

Amazon page purchased from: 
https://smile.amazon.com/gp/product/B07Q9YDS2V/ref=ppx_yo_dt_b_asin_title_o08_s03?ie=UTF8&psc=1

Can I use board ID "esp32dev" for the time being?

valeros, 2020-04-21:
Hi @jnr2820 ! It looks like it has the same pinout so you can use `esp32dev`.

jnr2820, 2020-04-25:
Awesome. Thanks, I'll try it out. I wasn't sure if it was significantly different from the "official" design.

closed 2020-05-25 by valeros