platformio/platform-espressif32 #330: Error: Couldn’t find the main target of the project!

howroyd, 2020-04-23:
As per my forum thread here:
https://community.platformio.org/t/error-couldnt-find-the-main-target-of-the-project/12773

Long time user of the ESP IDF, using their python scripts etc to build. Decided to make the leap to using platformio.

I have a project, which compiles fine from the command line with idf.py build but platformio is throwing me the error Error: Couldn't find the main target of the project!

The project structure is slightly different than the norm:

Inside the project dir, there is a folder called “core” which houses two other folders, “Src” and “Inc”. Where “Src” houses main.cpp. The “core” component is registered in the “CMakeLists.txt” file in this “core” folder.
Inside the project dir there is another folder called “application” which houses several folders which are setup as components, each with their own `CMakeLists.txt"
My platformio.ini is, and where I think my error must be:

```
[platformio]
src_dir = core/Src
[env]
platform = espressif32
framework = espidf
platform_packages = framework-espidf
[env:fm-devkit]
board = fm-devkit
```

And the terminal log:

```
Executing task in folder esp32: C:\Users\Simon.platformio\penv\Scripts\platformio.exe run <
Processing fm-devkit (board: fm-devkit; platform: espressif32; framework: espidf)
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/fm-devkit.html
PLATFORM: Espressif 32 1.12.0 > ESP32 FM DevKit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:

framework-espidf 3.40000.200303 (4.0.0)
tool-cmake 3.16.5
tool-esptoolpy 1.20600.0 (2.6.0)
tool-idf 1.0.1
tool-mconf 1.4060000.20190628 (406.0.0)
tool-ninja 1.9.0
toolchain-esp32ulp 1.22851.190618 (2.28.51)
toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration…
Error: Couldn’t find the main target of the project!
======================================================================================== [FAILED] Took 14.60 seconds ========================================================================================
The terminal process terminated with exit code: 1
Terminal will be reused by tasks, press any key to close it.
```

Many thanks for all your support, greatly appreciated.

closed 2020-04-27 by valeros

ya-mouse, 2022-03-01:
Check the glob wildcard in your `${src_dir}/CMakeLists.txt`, it should match your `${src_dir}`, e.g.:

**platformio.ini:**
```
[platformio]
src_dir = main
```

**main/CMakeLists.txt:**
```
FILE(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/main/*.*)
idf_component_register(SRCS ${app_sources})
```

Otherwise you will receive this error (component wouldn't registered with empty `SRCS`).

PS. It's also useful to check the Console output in `Help -> Toggle Developer Tools`

LEDFACIL, 2022-11-29:
> Check the glob wildcard in your `${src_dir}/CMakeLists.txt`, it should match your `${src_dir}`, e.g.:
>
> **platformio.ini:**
>
> ```
> [platformio]
> src_dir = main
> ```
>
> **main/CMakeLists.txt:**
>
> ``` [...]

Excellent thank you very much, I had the same problem, I did exactly that and it was solved, I simply did the following: in the platformio.ini file I added the line src_dir = main and that's it, solved! I didn't touch anything else, just added that line, attached screenshot:

[image]

catfreer, 2024-08-02:
before the ERROR ,i have a warning:Could not find "sdkconfig.json" file. Help!! i don't know how to solve the problem.

howroyd, 2024-08-02:
> before the ERROR ,i have a warning:Could not find "sdkconfig.json" file. Help!! i don't know how to solve the problem.

I recommend you ask on the forums or create a new issue with details on how to recreate the problem, like I did in my OP. Posting on an irrelevant and closed issue won't get you any help I'm afraid.

catfreer, 2024-08-02:
> I recommend you ask on the forums or create a new issue with details on how to recreate the problem, like I did in my OP. Posting on an irrelevant and closed issue won't get you any help I'm afraid.

Yeah, you‘re right, sorry to trouble you.

commit 9d5cc624 (2020-04-27): Dynamically detect the main component name // Resolve #330

CMake uses the folder name as component target