platformio/platform-espressif32 #331: Feature request: upload file system to a specific spiffs partition

mleusink, 2020-04-24:
When the user runs the 'PlatformIO: Upload File System Image' task, currently it is always uploaded to the last spiffs partition in the partition table.

For projects that contain multiple spiffs partitions, it would be nice to have some way for the user to specify which spiffs partition to upload the file system to.

It would be even better to support multiple data folders, with a way to specify which data folder should be uploaded to which spiffs partition.

For more detail about this request, please see the following forum discussion:
[https://community.platformio.org/t/upload-file-system-to-a-specific-spiffs-partition/12812](https://community.platformio.org/t/upload-file-system-to-a-specific-spiffs-partition/12812)

pfeerick, 2020-04-24:
Would the addition of another parameter, which lets you specify the data folder directory name do the trick? With the absence of the parameter defaulting to `data`, and presence of it would result in the configured directory used instead?

mleusink, 2020-04-24:
@pfeerick : That would help for the second part of this feature request.  But the main request is to be able to pick which spiffs partition to upload the file system to.

To clarify, let's spit the request into two parts:
1. If we have a single data folder and two spiffs partitions in the partition table, then I'm asking for a way to pick which partition the data folder gets uploaded to.

2. If we have two data folders and two spiffs partitions in the partition table, then I'm asking for a way to map each data folder to each spiffs partition.