platformio/platform-espressif32 #333: TTGO T-Call

MaksimKlyu, 2020-04-25:
https://github.com/platformio/platform-espressif32/issues/219

**pins_arduino.h**

#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>

#define EXTERNAL_NUM_INTERRUPTS 16
#define NUM_DIGITAL_PINS        20
#define NUM_ANALOG_INPUTS       16

#define analogInputToDigitalPin(p)  (((p)<20)?(esp32_adc2gpio[(p)]):-1)
#define digitalPinToInterrupt(p)    (((p)<40)?(p):-1)
#define digitalPinHasPWM(p)         (p < 34)

// Sim800
#define SIM800_TX 26
#define SIM800_RX 27
#define PWKEY 4
#define SIM800_RST 5
#define SIM800_POWER 23

static const uint8_t LED_BUILTIN = 13;
#define BUILTIN_LED  LED_BUILTIN // backward compatibility

static const uint8_t TX = 1;
static const uint8_t RX = 3;

static const uint8_t SDA = 21;
static const uint8_t SCL = 22;
 
static const uint8_t SS    = 18;
static const uint8_t MOSI  = 27;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 5;

static const uint8_t A0 = 36;
static const uint8_t A3 = 39;
static const uint8_t A4 = 32;
static const uint8_t A5 = 33;
static const uint8_t A6 = 34;
static const uint8_t A7 = 35;

static const uint8_t A10 = 4;
static const uint8_t A11 = 0;
static const uint8_t A12 = 2;
static const uint8_t A14 = 13;
static const uint8_t A16 = 14;
static const uint8_t A18 = 25;

static const uint8_t T0 = 4;
static const uint8_t T1 = 0;
static const uint8_t T2 = 2;
static const uint8_t T4 = 13;
static const uint8_t T6 = 14;
static const uint8_t T8 = 33;
static const uint8_t T9 = 32;
static const uint8_t DAC1 = 25;
#endif /* Pins_Arduino_h */

**ttgo-t-call.json**
{
  "build": {
    "core": "esp32",
    "extra_flags": "-DARDUINO_T_Call",
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "ldscript": "esp32_out.ld",
    "mcu": "esp32",
    "variant": "ttgo-t-call"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "can",
    "ethernet"
  ],
  "debug": {
    "openocd_board": "esp-wroom-32.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "TTGO T-CALL",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 512000,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://github.com/Xinyuan-LilyGO/TTGO-T-Call",
  "vendor": "TTGO"
}

ivankravets, 2020-04-25:
Could you make PR with `pins_arduino.h` to https://github.com/espressif/arduino-esp32/tree/master/variants ?

closed 2020-04-25 by ivankravets

MaksimKlyu, 2020-04-25:
https://github.com/MaksimKlyu/arduino-esp32/tree/patch-1/variants/ttgo-t-call