platformio/platform-espressif32 #334: Linking fails with latest arduino-esp32

margau, 2020-04-26:
Hello together,
with the latest arduino-esp32, I'm getting linker errors with platform-espressif32.
Compilation in Arduino with the same arduino-esp32 works fine (https://pastebin.com/Z6XaJUUz).
The errors with PIO: https://pastebin.com/q96pXzrm

The environment:
```
[env:demo_env]
platform = espressif32
board = esp-wrover-kit
upload_protocol = esptool
monitor_speed = 115200
framework = arduino
monitor_filers = esp32_exception_decoder
platform_packages =
      framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

OS is Linux Mint 19.3

This does not happen when using the default framework-arduinoespressif32. Unfortunately, due to #298 that's not a solution for me.

Best regards
margau

szechyjs, 2020-04-26:
I too am seeing this issue.

```
/Users/jared/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .dram0.data loaded at [0000000000000000,0000000000002e0f] overlaps section .iram0.vectors loaded at [0000000000000000,00000000000003ff]
/Users/jared/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .flash.rodata loaded at [0000000000000000,0000000000019eff] overlaps section .dram0.data loaded at [0000000000000000,0000000000002e0f]
/Users/jared/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .flash.text loaded at [0000000000000000,00000000000a9216] overlaps section .flash.rodata loaded at [0000000000000000,0000000000019eff]
/Users/jared/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: section .iram0.text loaded at [0000000000000400,0000000000014576] overlaps section .flash.text loaded at [0000000000000000,00000000000a9216]
...
vflash.c:(.text+0xc7): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: memset
vflash.c:(.text+0xf2): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: memcpy
/Users/jared/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/lib/libbtdm_app.a(vflash.o): In function `r_flash_write':
vflash.c:(.text+0x11e): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: memcpy
/Users/jared/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/lib/libbtdm_app.a(bt_util_buf.o): In function `r_bt_util_buf_init':
bt_util_buf.c:(.text+0x6f): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: memset
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

margau, 2020-04-27:
Good morning,
I have the suspection that the compiler is not called in the right direction when using the git version.
I compared the output of "pio run --verbose".
With "default" framework-arduinoespressif32 (successfull):
```
Scanning dependencies...
No dependencies
Building in release mode
xtensa-esp32-elf-g++ -o .pio/build/demo_env/src/main.cpp.o (...)
```
With git version (fails):
```
Scanning dependencies...
No dependencies
Building in release mode
xtensa-esp32-elf-g++ -o .pio/build/demo_env/firmware.elf -(...)
```

This seems like some build steps are missing when using the git version.

margau, 2020-04-27:
Some digging in the history suggests, that the problem maybe was introduced with 4c2b33894f70c941f34ab2bb55d778d2bee51886

margau, 2020-04-27:
I'm relatively sure to have found the problem:
The ldscript in the board.json-build-section moved from the global ldscript into the arduino-specific ldscript. For manually targeted git-checkouts, the "arduino"-section seems not to match. Manually adding `"ldscript": "esp32_out.ld",` into the json of the build board makes everything work fine.

A fix would probably contain the "arduino"-section of the build config also getting applied for arduino-git-checkouts. Trying to look into it.

margau, 2020-04-27:
Another Update: Workaround is adding `board_build.ldscript = esp32_out.ld` (or other ldscript) manually to the build-env - the root cause (config.build.arduino.ldscript not used when using framework-arduinoespressif32 as custom platform package) I haven't found yet.

@valeros, could you take a look at this part of configuration?
Thanks!

valeros, 2020-04-27:
Hi @margau ! There is a pending PR https://github.com/espressif/arduino-esp32/pull/3735 with changes required to work with the upstream Arduino for ESP32. You can try to add that two lines from the PR to the freshly pulled package and should work.

closed 2020-04-27 by valeros