platformio/platform-espressif32 #335: Setting CXX flags from extra_script.py doesn't work

DurandA, 2020-04-27:
I followed the [split C/C++ build flags](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#split-c-c-build-flags) instructions in order to use the `-fpermissive` flag when compiling .cpp files (`-fpermissive` is not valid for C).

I created a _extra_script.py_ file containing `env.Append(CXXFLAGS=["-fpermissive"])` and added `extra_scripts = extra_script.py` to _platformio.ini_. However, the flag is not used when building .cpp files (I have a _src_ folder with only _main.cpp_).

I did not test whether this issue is specifically related the _espressif32_ platform but most likely this is due to the ESP-IDF CMake build system of this platform.

valeros, 2020-04-27:
Hi @DurandA ! Thanks for the report. To correctly pass an additional build flag via an extra script you need to execute it before the main script of the development platform:
```
extra_scripts = pre:extra_script.py
```
Feel free to reopen the issue if you still have any problems.

closed 2020-04-27 by valeros

DurandA, 2020-04-28:
Many thanks!

KKoovalsky, 2020-05-06:
It actually doesn't work. With:

```
env.Append(
    CXXFLAGS=[
        '-Wall',
        '-Wextra',
        '-std=gnu++17',
        '-Wno-register'
    ]
)
```

1. Tried:
```
[env]
extra_scripts = pre:extra_script.py
```
2. Tried:
```
[env:native]
extra_scripts = pre:extra_script.py
```
Both doesn't work, but when I change `CXXFLAGS` to `CPPFLAGS` the flags are used BUT, both for `g++` and `gcc`.

valeros, 2020-05-06:
Hi @KKoovalsky ! Looks like you're having a problem with `native` platform. Please open an issue in this repository https://github.com/platformio/platform-native/issues
Thanks!

KKoovalsky, 2020-05-06:
@valeros  I can confirm that it's not only a problem for the `native` platform. It's a problem in general. It doesn't work for a NucleoL432KC board based environment.

Could you move the issue:
https://github.com/platformio/platform-native/issues/3