platformio/platform-espressif32 #339: Can't use build_flags in CMake condition.

saloid, 2020-04-30:
### Configuration

**Operating system**: Win 10 x64

**PlatformIO Version**: 4.3.3

### Description of problem
I am working with ESP32, ESP-IDF.
Flags which I define in build_flags can't be used in conditions in CMake.

#### Steps to Reproduce

1. Take any ESP-IDF project
2. Paste this in CMakeList.txt file in root of project
```cmake
IF(MY_TEST_FLAG)
  MESSAGE(FATAL_ERROR "TEST FLAG DEFINED")
ELSE()
  MESSAGE(FATAL_ERROR "TEST FLAG NOT DEFINED")
ENDIF()
```
3. In platformio.ini define this flag:
```
build_flags = -DMY_TEST_FLAG=ON
```
4. Clean, build
### Actual Results
```
CMake Error at CMakeLists.txt:14 (MESSAGE):
  TEST FLAG NOT DEFINED
```

### Expected Results
```
CMake Error at CMakeLists.txt:14 (MESSAGE):
  TEST FLAG DEFINED
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
src_dir = main

[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = espidf
upload_port = COM9
upload_speed = 2000000
monitor_port = COM9
monitor_speed = 115200
monitor_filters = colorize, esp32_exception_decoder, log2file

build_flags = -DMY_TEST_FLAG=ON

board_build.partitions = partition_table.csv
```

### Additional info
If I build project with native ESP-IDF toolchain (without platformio), it works as it should:
```
> python idf.py build -DMY_TEST_FLAG=ON
...
CMake Error at lib/httpd/CMakeLists.txt:14 (MESSAGE):
  TEST FLAG DEFINED
...
```

saloid, 2020-05-01:
Yep, it works, thanks. Will it be in the next release?

valeros, 2020-05-01:
Yes, it will.
Again, thanks for reporting. Closing this issue as resolved.

closed 2020-05-01 by valeros