platformio/platform-espressif32 #340: framework-arduinoespressif32: BLEClient::~BLEClient() crashes in BLERemoteService::removeCharacteristics() with double free

jackjansen, 2020-05-04:
I presume this issue really needs to be submitted in another repository, but all repository references for _framework-arduinoespressif32_ go to this repo. Please enlighten me if I need to submit it elsewhere. framework-arduinoespressif32 version is 3.10004.100129 (1.0.4).

I have found that creating a `BLEClient()` connection and then later calling `delete` on it is a sure way of getting a crash: `panic()` while freeing memory.

Digging down through the BLE stack showed that the double free is in `BLERemoteService::removeCharacteristics()`. It loops over two maps, `m_characteristicMap` and `m_characteristicMapByHandle`, calling `delete` on all the values.

*But*: the two maps share the same set of values (the characteristics objects), indexed by different keys. To fix, the delete loop should only be applied to `m_characteristicMap` and the other map should simply be cleared without deleting the values.

valeros, 2020-05-04:
Hi @jackjansen ! Thanks for the report. Could you please provide a minimal example to reproduce the issue?

jackjansen, 2020-05-05:
Owww, that is a difficult request, especially if it has to be a minimal _working_ example, because obviously I don't know what connectable BLE devices you will have available...

I will try to modify the Arduino BLE Client example to show the issue (but it may be a while).

valeros, 2020-05-05:
I'm mostly interested in your code, I didn't plan to run it on the real hardware.

jackjansen, 2020-05-06:
I _will_ try to write the sample code to demonstrate the bug, but in the mean time: I had also reported it initially in the platformio repo, https://github.com/espressif/arduino-esp32/issues/3966, and @jarikp has provided a fix that you can probably just pull over, at https://github.com/espressif/arduino-esp32/pull/3973

valeros, 2020-05-06:
Since there is a problem with the framework, you can manually patch your local framework package until that PR is merged into master.

jarikp, 2020-05-06:
Reproducing the problem which is fixed by https://github.com/espressif/arduino-esp32/pull/3973 is very easy, at least for me with ESP32 MCU. It simply requires a connection to a service with characteristics, so in the most trivial case the code will look like this:

```
#include <Arduino.h>
#include <BLEDevice.h>

void setup()
{  
    Serial.begin(115200);
    BLEDevice::init("");
}

void loop()
{
    BLEAdsress *pMacAdd = new BLEAddress("00:00:00:00:00"); // put a valid MAC here
    BLEClient *pClient = BLEDevice::createClient();
   
    pClient->connect(*pMacAdd , BLE_ADDR_TYPE_PUBLIC);

    BLERemoteService *pService = pClient->getService("00000000-0000-0000-0000-000000000000");
    BLERemoteCharacteristic *pCharacteristic = pService ->getCharacteristic("00000000-0000-0000-0000-000000000000");

    if (pCharacteristic->canRead()){
        Serial.println(pCharacteristic->readValue().c_str());
    }

    pClient->disconnect();
    delete pClient;

    delay(10000);
}
```

The expected behavior would be to read the value every 10th second, but the loop() crashes on "delete pClient" with this error message:

```
CORRUPT HEAP: Bad head at 0x3ffe0640. Expected 0xabba1234 got 0x3ffe0ba4
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400e6e73 on core 1

Backtrace: 0x40091470:0x3ffc84f0 0x400916a1:0x3ffc8510 0x400e6e73:0x3ffc8530 0x400910e5:0x3ffc8560 0x40084b26:0x3ffc8580 0x400850e1:0x3ffc85a0 0x4000bec7:0x3ffc85c0 0x4008dfa7:0x3ffc85e0 0x400d39af:0x3ffc8600 0x400d5aef:0x3ffc8620 0x400d6201:0x3ffc8640 0x400d657e:0x3ffc8660 0x400d659d:0x3ffc8680 0x400d5192:0x3ffc86a0 0x400d6852:0x3ffc86d0 0x400d1a7e:0x3ffc86f0 0x400d7bfd:0x3ffc8710 0x4008e089:0x3ffc8730
```

valeros, 2022-04-19:
Closing as the issue has nothing to do with the PlatformIO dev-platform.

closed 2022-04-19 by valeros